margelo/react-native-vision-camera #818: 🐛 takePhoto orientation always returns landscape asset on iOS

tedi, 2022-02-09:
### What were you trying to do?

I am trying to capture a photo with portrait orientation. The orientation prop is set to "portrait".

### Reproduceable Code

```tsx
<ReanimatedCamera
                      ref={camera}
                      style={StyleSheet.absoluteFill}
                      device={device}
                      format={format}
                      fps={fps}
                      hdr={enableHdr}
                      lowLightBoost={device.supportsLowLightBoost && enableNightMode}
                      isActive={isActive}
                      onInitialized={onInitialized}
                      onError={onError}
                      enableZoomGesture={true}
                      animatedProps={cameraAnimatedProps}
                      photo={true}
                      audio={true}
                      orientation="portrait"
                    />

const takePhotoOptions = useMemo<TakePhotoOptions & TakeSnapshotOptions>(
    () => ({
      photoCodec: 'jpeg',
      enableAutoStabilization: true,
      qualityPrioritization: 'balanced',
      flash: flash,
      skipMetadata: true,
    }),
    [flash],
  );

  const takePicture = useCallback(async () => {
    if (camera.current) {
      try {
        if (camera.current == null) throw new Error('Camera ref is null!');

        console.log('Taking photo...');
        const photo = await camera.current.takePhoto(takePhotoOptions);
        onPhotoCaptured(photo)
      } catch (e) {
        console.error('Failed to take photo!', e);
      }
    }
  }, [onPhotoCaptured, takePhotoOptions])

Also confirmed that the device is locked to portrait.
```

### What happened instead?

Media asset returned is in landscape mode

### Relevant log output

```shell
Xcode logs show everything is clear. This is the returned media (PhotoFile) obj:

{"height":2160,"metadata":{"{MakerApple}":{"1":14,"2":null,"3":{"timescale":1000000000,"epoch":0,"value":159656062542708,"flags":1},"4":1,"5":173,"6":176,"7":1,"8":[0.14260585606098175,-0.9722282290458679,-0.2048303186893463],"12":[0.1796875,0.2578125],"13":15,"14":0,"15":3,"16":1,"20":1,"23":0,"25":0,"31":0,"32":"BAB2A4BE-9E24-4F46-AC0D-DFF33150EA17","35":[362,268435524],"37":8,"38":1,"39":17.76178741455078,"43":"BEAAC845-CF70-4BB5-860A-DE9779A1477A","45":3896,"46":1,"47":58,"51":0,"52":1,"53":0,"54":90,"55":8,"58":0,"59":0,"60":4,"63":0,"65":false,"74":2},"{TIFF}":{"ResolutionUnit":2,"Software":"15.3","Make":"Apple","DateTime":"2022:02:09 18:48:01","XResolution":72,"HostComputer":"iPhone 12 Pro","Model":"iPhone 12 Pro","YResolution":72},"DPIWidth":72,"Orientation":6,"{Exif}":{"DateTimeOriginal":"2022:02:09 18:48:01","ExposureTime":0.016129032258064516,"FNumber":1.6,"LensSpecification":[4.2,4.2,1.6,1.6],"ExposureBiasValue":0,"ColorSpace":1,"FocalLenIn35mmFilm":29,"BrightnessValue":0.20465028074923952,"ExposureMode":0,"LensModel":"iPhone 12 Pro back camera 4.2mm f/1.6","SceneType":1,"PixelXDimension":3840,"ShutterSpeedValue":5.965603959052092,"SensingMethod":2,"SubjectArea":[1920,1077,2112,946],"ApertureValue":1.3561438102252754,"SubsecTimeDigitized":"492","FocalLength":4.2,"LensMake":"Apple","SubsecTimeOriginal":"492","OffsetTimeDigitized":"-05:00","PixelYDimension":2160,"ISOSpeedRatings":[640],"WhiteBalance":0,"DateTimeDigitized":"2022:02:09 18:48:01","OffsetTimeOriginal":"-05:00","ExifVersion":"0232","OffsetTime":"-05:00","Flash":16,"ExposureProgram":2,"MeteringMode":5},"DPIHeight":72},"isRawPhoto":false,"thumbnail":null,"path":"/private/var/mobile/Containers/Data/Application/FA4B93BB-2AD8-4CA0-96BE-94C2E83FE1F6/tmp/ReactNative/2E5F7B8A-9233-428E-8FB2-6B2358E08460.jpeg","width":3840}
```

### Device

iPhone 12 Pro

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-02-21:
Hey Tedi, unfortunately there's a lot of stuff happening at my agency so I didn't have any time yet to take a closer look at the issue you're experiencing. 

Maybe [the PR from @xulihang](https://github.com/mrousavy/react-native-vision-camera/pull/833) is related to the issue?

tedi, 2022-02-21:
I understand, thank you. I looked at that one, but it doesn't apply to iOS.

akshay8033, 2022-02-22:
+1 
Facing the same issue.

wego-deluxe, 2022-03-25:
Hi Folks,
any ideas for this issue? Facing the same problem on:

i(Pad)OS 14.4, 
XCode 12.4,
react-native-vision-camera 2.13.0,
react-native 0.67.3

In spite of device is in "portrait" position or setting vision camera property orientation="portrait" as shown in the description of @tedi above, the result image of takePhoto is always landscape - for example:
height: 2448, 
width: 3264, 
Orientation: 6

Also the metada, e.g. EXIF, is present although "skipMetadata: true" is set. 

I'm using the result image for text recognition with google ml-kit, so for now this does not work in portrait position of the device :(

mrousavy, 2022-03-28:
Maybe this is related to the orientation PR I merged a while ago, does it help if you revert that? My apps are using portrait only, so I can't really test this

tedi, 2022-03-28:
@mrousavy We're also only using portrait.

@wego-deluxe @akshay8033 we're doing a hacky workaround. We're cropping in the exact center of the returned image, with the size of the viewport, which essentially returns whatever is in the preview. We started working on a possible PR, but we're in the middle of a launch. I hope to return to this in 2-3 weeks. 

Regardless, this library is way ahead of anything else out there. Thank you @mrousavy for all your work.

mrousavy, 2022-03-29:
I see! Does it work if you revert https://github.com/mrousavy/react-native-vision-camera/commit/48da1819fc0fed894cd5c3ca994f9a5c93ed757b ?

akshay8033, 2022-03-29:
Our app allows capturing images in all orientations. We use device orientation and set the orientation. When the image is saved to the camera roll the orientation & dimensions are proper. 

The output of metadata I got in different orientations. 
Device - iPhone 12
```
PORTRAIT
{"height": 1080, "isRawPhoto": false, "metadata": {"DPIHeight": 72, "DPIWidth": 72, "Orientation": 6, "kCGImageDestinationICCProfile": null, "{Exif}": {"ApertureValue": 2.5260688116675873, "BrightnessValue": 0.7703026252288746, "ColorSpace": 65535, "CustomRendered": 1, "DateTimeDigitized": "2022:03:29 17:07:06", "DateTimeOriginal": "2022:03:29 17:07:06", "ExifVersion": "0232", "ExposureBiasValue": 0, "ExposureMode": 0, "ExposureProgram": 2, "ExposureTime": 0.02, "FNumber": 2.4, "Flash": 16, "FocalLenIn35mmFilm": 30, "FocalLength": 1.55, "ISOSpeedRatings": [Array], "LensMake": "Apple", "LensModel": "iPhone 12 back dual wide camera 1.55mm f/2.4", "LensSpecification": [Array], "MeteringMode": 5, "OffsetTime": "+05:30", "OffsetTimeDigitized": "+05:30", "OffsetTimeOriginal": "+05:30", "PixelXDimension": 1920, "PixelYDimension": 1080, "SceneType": 1, "SensingMethod": 2, "ShutterSpeedValue": 5.6442890632212555, "SubjectArea": [Array], "SubsecTimeDigitized": "469", "SubsecTimeOriginal": "469", "WhiteBalance": 0}, "{MakerApple}": {"1": 14, "14": 0, "2": null, "20": 1, "23": 1107329152, "25": 0, "3": [Object], "31": 0, "32": "24CC74C7-22DB-4FEC-B009-EFDBAF804521", "37": 524, "38": 2, "39": 19.29308319091797, "4": 1, "43": "23EF33EB-D0E5-4BE1-96FC-CF2C92899E1D", "45": 6809, "46": 0, "47": 0, "5": 171, "51": 0, "52": 1, "53": 0, "54": 116, "55": 0, "58": 4, "59": 1, "6": 173, "60": 16, "63": 0, "65": false, "7": 1, "74": 4, "8": [Array]}, "{TIFF}": {"DateTime": "2022:03:29 17:07:06", "HostComputer": "iPhone 12", "Make": "Apple", "Model": "iPhone 12", "ResolutionUnit": 2, "Software": "15.3.1", "XResolution": 72, "YResolution": 72}}, "path": "/private/var/mobile/Containers/Data/Application/08784C48-C80B-4D58-8B6D-F0529AA443D3/tmp/ReactNative/2AC12FDC-8042-47F5-8B1A-2DBA778C5180.jpeg", "thumbnail": null, "width": 1920}

LANDSCAPE-LEFT
{"height": 1080, "isRawPhoto": false, "metadata": {"DPIHeight": 72, "DPIWidth": 72, "Orientation": 1, "kCGImageDestinationICCProfile": null, "{Exif}": {"ApertureValue": 2.5260688116675873, "BrightnessValue": 0.12137458553476134, "ColorSpace": 65535, "CustomRendered": 1, "DateTimeDigitized": "2022:03:29 17:07:25", "DateTimeOriginal": "2022:03:29 17:07:25", "ExifVersion": "0232", "ExposureBiasValue": 0, "ExposureMode": 0, "ExposureProgram": 2, "ExposureTime": 0.02, "FNumber": 2.4, "Flash": 16, "FocalLenIn35mmFilm": 30, "FocalLength": 1.55, "ISOSpeedRatings": [Array], "LensMake": "Apple", "LensModel": "iPhone 12 back dual wide camera 1.55mm f/2.4", "LensSpecification": [Array], "MeteringMode": 5, "OffsetTime": "+05:30", "OffsetTimeDigitized": "+05:30", "OffsetTimeOriginal": "+05:30", "PixelXDimension": 1920, "PixelYDimension": 1080, "SceneType": 1, "SensingMethod": 2, "ShutterSpeedValue": 5.6442890632212555, "SubjectArea": [Array], "SubsecTimeDigitized": "733", "SubsecTimeOriginal": "733", "WhiteBalance": 0}, "{MakerApple}": {"1": 14, "14": 0, "2": null, "20": 1, "23": 1107329152, "25": 0, "3": [Object], "31": 0, "32": "7EC8D308-1F5E-4A3A-A080-E484285914BB", "37": 524, "38": 2, "39": 17.137588500976562, "4": 1, "43": "5057A47D-4287-4845-B9C8-F1AB9998A6E0", "45": 6517, "46": 0, "47": 0, "5": 169, "51": 0, "52": 1, "53": 0, "54": 70, "55": 0, "58": 4, "59": 1, "6": 172, "60": 16, "63": 0, "65": false, "7": 1, "74": 4, "8": [Array]}, "{TIFF}": {"DateTime": "2022:03:29 17:07:25", "HostComputer": "iPhone 12", "Make": "Apple", "Model": "iPhone 12", "ResolutionUnit": 2, "Software": "15.3.1", "XResolution": 72, "YResolution": 72}}, "path": "/private/var/mobile/Containers/Data/Application/08784C48-C80B-4D58-8B6D-F0529AA443D3/tmp/ReactNative/C8F6CA33-12D4-4C10-8800-A59EB0DD661C.jpeg", "thumbnail": null, "width": 1920}

LANDSCAPE-RIGHT
{"height": 1080, "isRawPhoto": false, "metadata": {"DPIHeight": 72, "DPIWidth": 72, "Orientation": 3, "kCGImageDestinationICCProfile": null, "{Exif}": {"ApertureValue": 2.5260688116675873, "BrightnessValue": -0.37259434283534876, "ColorSpace": 65535, "CustomRendered": 1, "DateTimeDigitized": "2022:03:29 17:11:08", "DateTimeOriginal": "2022:03:29 17:11:08", "ExifVersion": "0232", "ExposureBiasValue": 0, "ExposureMode": 0, "ExposureProgram": 2, "ExposureTime": 0.02, "FNumber": 2.4, "Flash": 16, "FocalLenIn35mmFilm": 30, "FocalLength": 1.55, "ISOSpeedRatings": [Array], "LensMake": "Apple", "LensModel": "iPhone 12 back dual wide camera 1.55mm f/2.4", "LensSpecification": [Array], "MeteringMode": 5, "OffsetTime": "+05:30", "OffsetTimeDigitized": "+05:30", "OffsetTimeOriginal": "+05:30", "PixelXDimension": 1920, "PixelYDimension": 1080, "SceneType": 1, "SensingMethod": 2, "ShutterSpeedValue": 5.6442890632212555, "SubjectArea": [Array], "SubsecTimeDigitized": "494", "SubsecTimeOriginal": "494", "WhiteBalance": 0}, "{MakerApple}": {"1": 14, "14": 0, "2": null, "20": 1, "23": 1107329152, "25": 0, "3": [Object], "31": 0, "32": "D75C04D4-11BF-4E1E-8E1D-6A023209789F", "37": 524, "38": 2, "39": 15.433605194091797, "4": 1, "43": "54E98889-E56E-44C9-BC85-61ACCF0EE437", "45": 6587, "46": 0, "47": 0, "5": 162, "51": 0, "52": 1, "53": 0, "54": 53, "55": 0, "58": 4, "59": 1, "6": 171, "60": 16, "63": 0, "65": false, "7": 1, "74": 4, "8": [Array]}, "{TIFF}": {"DateTime": "2022:03:29 17:11:08", "HostComputer": "iPhone 12", "Make": "Apple", "Model": "iPhone 12", "ResolutionUnit": 2, "Software": "15.3.1", "XResolution": 72, "YResolution": 72}}, "path": "/private/var/mobile/Containers/Data/Application/08784C48-C80B-4D58-8B6D-F0529AA443D3/tmp/ReactNative/A9FC1A21-A226-4F76-8275-FC10079A99AE.jpeg", "thumbnail": null, "width": 1920}
```

ethul, 2022-03-29:
Just wanted to note that I have the same issue and tested on version 2.11.1, before #715, and still have the issue that the photo result is always in landscape orientation on iOS.

I have experienced that both photo and video are always landscape on an iPhone 5s, and that photo is always landscape on an iPhone 8, but video does result in portrait orientation on an iPhone 8. Both 2.11.1 and the latest release produce these results on the iPhone 5s and the iPhone 8.

mrousavy, 2022-03-30:
This seems like it requires lots of efforts to investigate and good test results with a lot of different edge cases.

Unfortunately I don't have enough free time to focus on that, and I prioritize other, maybe even more critical issues (with easier fixes) over this one.

If you want this issue to get fixed, you can help by creating a PR, adding good reproduction steps to cover all different cases, and fix it yourself. Alternatively, if you're a company contact me for enterprise support and we (my app development agency Margelo) can work on this issue to get it solved in no time.

nguyenxdat, 2022-03-31:
@tedi @ethul @mrousavy 

In the [Apple Document](https://developer.apple.com/documentation/uikit/uiimage/orientation) there is talk about this issue. 

`For example, an iOS device camera always encodes pixel data in the camera sensor's native landscape orientation, along with metadata indicating the camera orientation. When UIImage loads a photo shot in portrait orientation, it automatically applies a 90° rotation before displaying the image data, and the image's [imageOrientation](https://developer.apple.com/documentation/uikit/uiimage/1624141-imageorientation) value of [UIImage.Orientation.right](https://developer.apple.com/documentation/uikit/uiimage/orientation/right) indicates that this rotation has been applied.`

And this is how we get [height](https://github.com/mrousavy/react-native-vision-camera/blob/main/ios/PhotoCaptureDelegate.swift#L50) and [width](https://github.com/mrousavy/react-native-vision-camera/blob/main/ios/PhotoCaptureDelegate.swift#L49) of the image. 

I think we need change logic in here, look like this:

[image]

ethul, 2022-03-31:
@nguyenxdat Thanks so much for this info! I will give the change you suggest a try.

karamanmertcan, 2022-04-21:
Guys, android has this problem also. It gives the height and width reverse if you use takePhoto function. It is giving the correct results with takeSnaphot function . Do you have any solution for that ?

ethul, 2022-04-21:
I had to make a few changes for it to work in my case:
https://github.com/ethul/react-native-vision-camera/commit/404f2311d807e33e0477dfac022b18312b2b20c4

karamanmertcan, 2022-04-21:
@ethul  Thanks, your solution was very helpful 👍

bryanltobing, 2023-06-19:
> I had to make a few changes for it to work in my case:
https://github.com/ethul/react-native-vision-camera/commit/404f2311d807e33e0477dfac022b18312b2b20c4

is this fixed? or we still need to use this fork?

closed 2023-09-30 by mrousavy