margelo/react-native-vision-camera #821: How do you call function to parent component from frame processors❓

ademirmochi, 2022-02-10:
### Question

Hi there, I'm kinda new to React Native, it might actually be a stupid question... If it is, sorry...
So I've been having an issue with react-native-vision-camera frame processors, and I used Image-labeler as well to make an object identification app, fairly simple, the problem is, since the camera component is a child component of my main component, I tried to check whether the thing I'm looking at is precisely what I want, and if so, call a parent function to change screens.

So... how can I do that? Call a parent component function from within frameprocessors?

### What I tried

  ```
  const frameProcessor = useFrameProcessor((frame) => {
      'worklet';
      const labels = labelImage(frame);
      if(labels[0]?.label != undefined)
      {
        currentLabel.value = labelIdentificada;
      }
      console.log(currentLabel.value)
      if(currentLabel.value === "Thing I want")
      {
        console.log('got in')
        props.onCallNewScreen();
      }
    }, [currentLabel])
```
Something like this, but the app simply crashes without any warning and/or errors, after I while trying different solutions, I realized that, even if I call another function in the same component, the system crashes.
I don't know if this is a frame processors problem, I think not since this seems like basic hooks issue and lack of information from me in how to proceed, since I'm new to this, but at this point, any help is appreciated.

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-02-11:
Hi! This is based off of Reanimated Worklets, so to call a normal JS function you need to use `runOnJS(props.onCallNewScreen)()`

closed 2022-02-11 by mrousavy

ademirmochi, 2022-02-11:
Dude, that's precisely it, and it was my lack of attention because it seems that this information was already in the frame processors docs, sorry to take your time, and thank you so much for your help!

xulihang, 2022-02-16:
Is there a way to get the returned value of the function?

mrousavy, 2022-02-16:
Wdym? Yes, it's simple:

```ts
const frameProcessor = useFrameProcessor((frame) => {
  'worklet'
  const faces = detectFaces(frame)
  console.log(faces)
}, [])

```

xulihang, 2022-02-16:
I mean get the function result of `runOnJS`.

Currently, I am using useSharedValue. I run the function in the parent component, update the shared value and then access it in the frame processor.

mrousavy, 2022-02-16:
Ah, no this is not possible due to limitations from Reanimated. I am working on an alternative.