margelo/react-native-vision-camera #822: When the camera view is unmounted with frameProcessor enabled, there was a view not found error

xulihang, 2022-02-12:
### What were you trying to do?

I use react navigation to open a barcode scanning page. After I pressed the back button on Android, there was the ViewNotFoundError.

### What happened instead?

A ViewNotFoundError pops up.

### Device

Sharp Aquos S2 (Android 8)

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

xulihang, 2022-02-12:
All right, there are similar issues: https://github.com/mrousavy/react-native-vision-camera/issues/626#issuecomment-1004699011

xulihang, 2022-02-12:
Hiding the header and using the back button can avoid the error from happening.

```js
<Root.Group screenOptions={{ headerShown: false }}>
  <Root.Screen name="BarcodeScanner" component={BarcodeScanner} />
</Root.Group>
```

digitalbore, 2022-02-14:
My header is disabled (headerShown: false) and I get the same error.

This is how my App is composed.

```
   <Tab.Navigator
      screenOptions={{
        headerShown: false,
        tabBarShowLabel: false,
      }}>
      <Tab.Screen
        name={i18n.t('navigation.finder')}
        component={List}
      />
      <Tab.Screen
        name={i18n.t('navigation.finder')}
        component={Finder}
      />
      <Tab.Screen
        name={i18n.t('navigation.more')}
        component={More}
      />
    </Tab.Navigator>
```

The Camera is mounted in the central tab (Finder), I get the error randomly when I switch between these tabs.

xulihang, 2022-02-15:
I don't have an Android 11 device. You can try this example: https://github.com/xulihang/react-native-barcode-overlays

xulihang, 2022-02-17:
I am also getting this error: `Can't perform a React state update on an unmounted component.`

It seems that the frame processor is still running and performing a React state update which I write in the processor.

I am trying to avoid this with this commit: https://github.com/xulihang/react-native-barcode-overlays/commit/81d6c81b93875fd3704c9ae3ec27e7fae74d761f

But the warning will come up occasionally.

mrousavy, 2022-03-30:
Hey! Does [v2.13.1](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.13.1) fix the issue for you?

fabio-C, 2022-03-30:
Yes, it works! Thanks @mrousavy