margelo/react-native-vision-camera #824: How do I send the photo using fetch❓

ToasterIntervention, 2022-02-13:
### Question

How would I use fetch to POST a photo to my backend? I am not sure how to convert the file name file:///...name.jpg to a format that can be sent. I am fairly new at this, so it would be greatly helpful if anyone can post a working code for the file processing and POST request or provide more details suitable for a noob. Any help would be greatly appreciated!  I know that the author of this library is working on a faster way to save files, but unfortunately my school project is due within the week. ~ cry

(I am not using Expo)

### What I tried

I tried saving the file as a blob, but never was successful. I also tried to send it directly to the backend but the file was always empty.

### VisionCamera Version

Latest version

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

ToasterIntervention, 2022-02-14:
I managed to access the file in cache with react-native-fs

closed 2022-02-14 by ToasterIntervention

mrousavy, 2022-02-14:
Hey! Good luck with the school project 😄

JosMMV, 2022-06-08:
Hi @ToasterIntervention, could you share the way you managed it with react-native-fs?