margelo/react-native-vision-camera #830: android\build is cleaned after a new npm package is installed❓

xulihang, 2022-02-15:
### Question

Every time I install a new package, the build folder (`node_modules\react-native-vision-camera\android\build`) is cleaned and it has to download packages like boost, folly again. I wonder if there is a way to keep the folder.

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-02-15:
Hi!

Yes, this is because I have to get boost and folly from somewhere. Those packages are served over a web server and since C++ does not have a package manager like NPM, I cannot install them like this, I have to manually download them.

The solution would be **Android prefabs**, that requires React Native to ship their binaries as a prefab.

closed 2022-02-15 by mrousavy

cognvn, 2022-02-18:
Hi @mrousavy, how about changing `downloadsDir` here
https://github.com/mrousavy/react-native-vision-camera/blob/06bcb53691c5029b08285b434e67c68ad723c0b7/android/build.gradle#L265
to
```diff
  def downloadsDir = new File("$buildDir/downloads")
+ if (findProject(':app') != null) {
+   downloadsDir = new File("${project(':app').buildDir}/downloads")
+ }
  def thirdPartyNdkDir = new File("$buildDir/third-party-ndk")
```
Apply these changes to `react-native-mmkv` and `react-native-multithreading`, we can use those packages for them without repeating download.

cognvn, 2022-02-18:
Sorry, I'm not sure what you mean by '**versioned**' :worried: