margelo/react-native-vision-camera #834: 🐛  Example app broken when switching devices

AdamTyler, 2022-02-17:
### What were you trying to do?

Switch devices while recording. Is this a supported feature? 

Its not working in my own app so I tried to reproduce in example app and see the same issue.

### Reproduceable Code

```tsx
Example app. bootstrapped and running android.
```

### What happened instead?

Upon flipping the camera device between front or back when recording (double tap on the screen), an error message comes up and the recording will not finish properly

### Relevant log output

```shell
LOG  Re-rendering Navigator. Camera: authorized | Microphone: authorized
 LOG  re-rendering camera page without active camera
 LOG  re-rendering camera page without active camera
 LOG  Re-rendering camera page with active camera. Device: "back (0)" (4032x1816 @ 30fps)
 LOG  Camera initialized!
 LOG  Re-rendering camera page with active camera. Device: "back (0)" (4032x1816 @ 30fps)
 DEBUG  state: BEGAN
 LOG  calling startRecording()...
 LOG  called startRecording()!
 LOG  Re-rendering camera page with active camera. Device: "front (1)" (3648x1640 @ 30fps)
 ERROR  [device/invalid-device: [device/invalid-device] The given Camera device could not be found for use-case binding!]
 ERROR  Recording failed! {"cause": {"message": "[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.", "stacktrace": "com.mrousavy.camera.InactiveSourceError: [capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.
	at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:64)
	at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:52)
	at androidx.camera.video.Recorder$RecordingRecord.lambda$updateVideoRecordEvent$6$Recorder$RecordingRecord(Recorder.java:2611)
	at androidx.camera.video.-$$Lambda$Recorder$RecordingRecord$Uo40MCKPuma4aZ5df1-_fzufzHg.run(Unknown Source:4)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.app.ActivityThread.main(ActivityThread.java:8641)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:567)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1133)
"}, "code": "capture/inactive-source", "message": "[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.", "userInfo": null}
 LOG  stopped recording video!
 DEBUG  state: ACTIVE
 DEBUG  state: END
 LOG  calling stopRecording()...
 LOG  called stopRecording()!
```

### Device

Samsung Galaxy S20+

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-02-17:
Hey, thanks for the bug report! I think on iOS this works, but on Android it does not work because the native camera library (CameraX) does not support this. I have created an issue in their issuetracker, no response yet.

AdamTyler, 2022-02-17:
@mrousavy thanks for the quick response! 
I went ahead and verified that it does work on iOS which is awesome.
One thing I do notice though is a quick flash of the wrong orientation stretched to fit when switching between them that shows up in the final recording.

Hopefully CameraX can add support soon!

AdamTyler, 2022-02-17:
@mrousavy here is an example of how it’s recording 

https://user-images.githubusercontent.com/2525253/154533243-8ce426c9-886d-4c61-b2df-ba1037413d9f.MOV

mrousavy, 2022-02-18:
Interesting! This might be a bug due to the orientation change I have added?

usaidather, 2022-03-28:
@mrousavy any update on this issue ? because I am also facing the same issue.

roger10cassares, 2022-05-11:
Up+

Aakash359, 2022-10-12:
The recording failed because the source becomes inactive and stops sending frames. One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated...... gtting error in android, this is perfectly working in iOS but not in android can any tell me what the issues is there

zzzkasper-1995, 2022-10-14:
@AdamTyler I have same issue with stretch and orientation

ratz6, 2023-04-15:
Any fix for the Video being stretched while switching cameras on iOS ?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy