margelo/react-native-vision-camera #835: ❓ Any way to load raw assets from frame processor plugin package?

pot2mayo, 2022-02-18:
### Question

Hi

The frame processor plugin system is very neat and lots of nice samples are provided.

I was wondering if there is any way to load raw assets (eg. AI models) to be used within the plugin.

I checked the MLKit-based samples but those act as a blackbox and we cannot see how models are loaded internally.

Thanks!

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

pot2mayo, 2022-02-18:
@mrousavy 
Thanks for the quick reply!

Drawing back to the cam is ok, since I'm only planning to return the result of the AI models as an array of bounding boxes, and from I could see this is possible. Then I can draw the bounding boxes on the RN side.

My plan is to use ONNX Runtime for the inference (something like [this](https://github.com/microsoft/onnxruntime-inference-examples/tree/main/mobile/examples/image_classification/android)), and for that I need to load the ORT model (like [this](https://github.com/microsoft/onnxruntime-inference-examples/blob/main/mobile/examples/image_classification/android/app/src/main/java/ai/onnxruntime/example/imageclassifier/MainActivity.kt#L153)) but I could not get to load the raw asset from within the frame processor plugin.

Any idea if this is doable?

mrousavy, 2022-02-18:
I have no experience with ONNX or ORT, but yes that should be possible just like in a native app :)

pot2mayo, 2022-02-20:
@mrousavy 
Since I've been stuck on this for 2 days now, please let me ask one more question.

Have you ever managed to open a raw resource (anything, like a text file in the `res/raw` folder) from within the frame processor plugin code?

There might be some kind of limitation preventing me from doing it, not sure.

If I could do that I think it would unlock everything.

Thanks!

mrousavy, 2022-02-20:
Hi @pot2mayo, no I have not tried to do that yet. But it is possible to do, because your Frame Processor Plugin is native code. You can do everything you want to do in a native app, including asset loading.

pot2mayo, 2022-02-21:
@mrousavy 
I agree, it should be possible but unfortunately with raw resources (`res/raw`) as well as asset resources (`assets` folder) I get various errors.
It's like the way the frame processor plugin is embedded in the whole project does not allow resource access :(

pot2mayo, 2022-02-21:
@mrousavy 
Nevermind I think I found a way.
It seems like accessing assets via the ReactPackage passed to the package constructor allows me to access both raw resources and assets.

For reference is someone runs in the same issue, the code below seems to open the raw assets correctly from both `R.raw` and `assets` (from both for testing but you most likely need only one way to load your asset).

```
  public List<NativeModule> createNativeModules(@NonNull @org.jetbrains.annotations.NotNull ReactApplicationContext reactContext) {
    int modelID = R.raw.myfile;
    String modelFilename = "myfile";
    InputStream is = null;

    try {
      Log.v(TAG, "Opening asset resource " + modelFilename);
      AssetManager assetManager = reactContext.getAssets();
      is = assetManager.open(modelFilename);
      Log.v(TAG, "Opening asset resource successfully. Size=" + is.available());

      Log.v(TAG, "Opening raw resource " + modelID);
      is = reactContext.getResources().openRawResource(modelID);
      Log.v(TAG, "Open raw resource " + modelID + ". Size=" + is.available());
      byte[] b = new byte[is.available()];
      is.read(b);
    } catch (Exception e) {
      // e.printStackTrace();
      Log.e(TAG, "Error: failed to open raw resource " + modelID + ". " + e.getLocalizedMessage());
    }
```

closed 2022-02-21 by pot2mayo

mrousavy, 2022-02-21:
Nice work!