margelo/react-native-vision-camera #843: ❓How to scan GS1-DataBar Expanded code?

SarotaM, 2022-02-22:
### Question

How can i scan and decrypt GS1-DataBar Expanded code with react-native-vision-camera? It is even possible to do it?

### What I tried

I have tried to use 'vision-camera-code-scanner' package in react-native app, but this package doesn't recognize this type of code.

### VisionCamera Version

2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-02-22:
Hey, you should use Frame Processors for this. If the existing packages do not support this specific code yet, you can easily build your own Frame Processor Plugin :)

closed 2022-02-22 by mrousavy

xulihang, 2022-02-23:
You can try [vision-camera-dynamsoft-barcode-reader](https://github.com/xulihang/vision-camera-dynamsoft-barcode-reader) which supports GS1-DataBar Expanded.