margelo/react-native-vision-camera #854: 🐛 [unknown/unknown] Value for flash cannot be cast from Boolean to String

Real-Musafir, 2022-02-28:
### What were you trying to do?

Was trying to take a photo in android  11

### Reproduceable Code

```tsx
<Camera
						style={styles.cameraContainer}
						ref={camera}
						device={device}
						isActive={true}
						photo={true}
						torch={'off'}
					/>
```

### What happened instead?

but was greeted with this error in metro :

🐛 [unknown/unknown] Value for flash cannot be cast from Boolean to String

### Relevant log output

```shell
Possible Unhandled Promise Rejection (id: 0):
unknown/unknown: [unknown/unknown] Value for flash cannot be cast from Boolean to String
construct@[native code]
Wrapper@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.basementsports&modulesOnly=false&runModule=true:7135:61
construct@[native code]
```

### Device

android 11 realme 8

### VisionCamera Version

"^2.12.0"

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-02-28:
You are passing a wrong value to the `flash` parameter. Also you didn't share your code, only the camera view.

closed 2022-02-28 by mrousavy

kaiyes, 2022-05-11:
Take out that torch={'off'} from camera component & then you can either add it in the `camera.current.takePhoto` code or omit it so that default is taken

[image]
[image]