margelo/react-native-vision-camera #855: Camera is showing white screen instead of showing camera in android🐛

humza-reignsol, 2022-02-28:
### What were you trying to do?

when I loaded the camera screen it is just showing white blank page.
if I resize camera, it start showing the actual camera screen but whenever I try to give width: ('Dimensions').width, height: ('Dimensions').height camera page is blank once again.
Flash is working fine but don't know why camera page is showing white blank page.
I'm attaching a screenshot below: 

[image]

### Reproduceable Code

```tsx
<Reanimated.View style={StyleSheet.absoluteFill}>
            <TapGestureHandler onEnded={onDoubleTap} numberOfTaps={2}>
              <ReanimatedCamera
                ref={camera}
                style={StyleSheet.absoluteFill}
                device={device}
                format={format}
                fps={fps}
                hdr={enableHdr}
                lowLightBoost={device.supportsLowLightBoost && enableNightMode}
                isActive={isActive}
                onInitialized={onInitialized}
                onError={onError}
                enableZoomGesture={false}
                animatedProps={cameraAnimatedProps}
                photo={true}
                video={false}
                audio={hasMicrophonePermission}
                orientation="portrait"
              />
            </TapGestureHandler>
          </Reanimated.View>
```

### What happened instead?

I have pasted the code I'm using.
Kindly correct me if I'm wrong somewhere. 

### Device

OPPO F-19, Techno Camon16

### VisionCamera Version

^2.12.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-02-28:
Hmm, are you running the example app without making any modifications?

humza-reignsol, 2022-02-28:
> Hmm, are you running the example app without making any modifications?

Pardon, it was an issue on my side.
I was trying to render camera in child view which caused this issue.

nachoSource, 2022-07-22:
Well this is curious 😅 . Your answer helped me realise that `Camera` must be wrapped in `<></>` instead of a `View` component 🤔

Brittognanaraj, 2023-02-08:
Try this one android:hardwareAccelerated="false" in AndroidManifest.xml makes the camera no longer display. Check your AndroidManifest and set it back to true may help the camera work again.

mrousavy, 2023-02-14:
Ohh yea maybe the SurfaceView doesn't work then?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy