margelo/react-native-vision-camera #862: Can we create a record from frameProcessor (record video with a watermark / label)❓

wibimaster, 2022-03-02:
### Question

Hi !

Thanks for this awesome lib', but there is a thing I want to do, and I'm not sure that it can answer to :

**While recording a video, I want to add variable texts on it (like a label at the bottom).**

With frameProcessor, I think I can read my RN states and add a little text box, but it seems to live aside the <Camera> recording, not in it...

1. Does this plugin permit to record a video _WITH_ the labels on the screen ?
2. Or is there a way to create the mp4 directly from the frameProcessor, that receives the video frame + the RN state and can update each frame with the label box on it ?

Thanks !

### VisionCamera Version

2.12.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-03-02:
Hi!

No, frame processors are currently "read only"/"blackbox", and everything you draw onto the Camera View is not part of the hardware capture pipeline.

This would be possible by simulatneously recording the video as usually (mp4), and then drawing the rendered content (your watermark) into the video before writing it each frame. This is how ARKit (Swift/iOS) does this as well, but it's really really complex.

wibimaster, 2022-03-02:
Hi, 

Thanks for the answer,

I take a look at [react-native-video-watermark](https://github.com/natibekele/react-native-video-watermark), which basically use (for Android) [mp4compose](https://github.com/MasayukiSuda/Mp4Composer-android) ; it seems to replay the video, and draw a canvas for each frame, applying on it various filters.

In my case, I don't need real time preview, but I need a watermark-like that can change while video running (unlike mp4compose).

It seems that no solution exists at this time :/

That could be a potential awesome feature for this plugin ;)

IamHaseebAhmed, 2022-03-10:
Hi, maybe you can implement watermark like any image layout inside the video using the ffmpeg library.

thomas-coldwell, 2022-10-22:
Marking this as a duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/724. We are actively working on adding support for write-back frame processors in Vision Camera V3 (the in-progress PR for this functionality can be found at https://github.com/mrousavy/react-native-vision-camera/pull/1198).

closed 2022-10-22 by thomas-coldwell

fukemy, 2023-06-09:
@thomas-coldwell , Can u help? I can not install #1198 pull request to check with you