margelo/react-native-vision-camera #866: 🐛 frame-processor/unavailable: Frame Processors are not enabled

msnainps, 2022-03-04:
### What were you trying to do?

I want to attach a frame processor. But not working on android 7.

### Reproduceable Code

```tsx
Just try to add a frame processor on android 7 and it will crash the app with the message Frame Processors are not enabled.
```

### What happened instead?

frame-processor/unavailable: Frame Processors are not enabled. See https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting

### Device

Motorola  (Android 7.0)

### VisionCamera Version

2.12.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-03-04:
Did you install Reanimated v2?

msnainps, 2022-03-04:
@mrousavy  Yes, I am using "react-native-reanimated": "^2.4.1",

wouterds, 2022-03-06:
Same issue here, fresh RN project with iOS. Followed setup of react-native-reanimated & react-native-vision-camera.

kofkgoing, 2022-03-15:
https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting

8. If your Frame Processor is not running, make sure you check the native Android Studio/Logcat logs to find out why. Also make sure you are not using a remote JS debugger such as Google Chrome, since those don't work with JSI.

mrousavy, 2022-03-15:
Exactly, thanks @kofkgoing.

Try deleting all cache and rebuilding, I think this is a caching issue.

closed 2022-03-15 by mrousavy