margelo/react-native-vision-camera #867: ❓ iPhone 11 Pro - No slow-motion formats available on back camera

dyelax, 2022-03-05:
### Question

I'm experimenting on my iPhone 11 Pro, which should have a 240 fps option for the back camera (according to [the spec](https://support.apple.com/kb/SP805?locale=en_US)). However, when I list the available formats, all I see are 30fps and 60fps framerate options for the back camera. When I list options for the front camera, I see some 120fps options, but I'd like to be able to record with the back camera for the higher framerate and wider field of view. 

Any ideas why the 240fps formats might not be showing?

### devices.back.formats:
```
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb"], "fieldOfView": 100.46436309814453, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 189, "minISO": 32, "photoHeight": 3024, "photoWidth": 4032, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 144, "videoStabilizationModes": ["auto", "off"], "videoWidth": 192}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb", "p3-d65"], "fieldOfView": 100.46436309814453, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 189, "minISO": 32, "photoHeight": 3024, "photoWidth": 4032, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 144, "videoStabilizationModes": ["auto", "off"], "videoWidth": 192}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb"], "fieldOfView": 92.09233093261719, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 189, "minISO": 32, "photoHeight": 3024, "photoWidth": 3696, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 288, "videoStabilizationModes": ["auto", "off"], "videoWidth": 352}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb", "p3-d65"], "fieldOfView": 92.09233093261719, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 189, "minISO": 32, "photoHeight": 3024, "photoWidth": 3696, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 288, "videoStabilizationModes": ["auto", "off"], "videoWidth": 352}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb"], "fieldOfView": 100.46436309814453, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 189, "minISO": 32, "photoHeight": 3024, "photoWidth": 4032, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 360, "videoStabilizationModes": ["auto", "off"], "videoWidth": 480}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb", "p3-d65"], "fieldOfView": 100.46436309814453, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 189, "minISO": 32, "photoHeight": 3024, "photoWidth": 4032, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 360, "videoStabilizationModes": ["auto", "off"], "videoWidth": 480}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb"], "fieldOfView": 100.46436309814453, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 189, "minISO": 32, "photoHeight": 3024, "photoWidth": 4032, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 480, "videoStabilizationModes": ["auto", "off"], "videoWidth": 640}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb", "p3-d65"], "fieldOfView": 100.46436309814453, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 189, "minISO": 32, "photoHeight": 3024, "photoWidth": 4032, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 480, "videoStabilizationModes": ["auto", "off"], "videoWidth": 640}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb"], "fieldOfView": 100.46436309814453, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 2}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 94.5, "minISO": 32, "photoHeight": 1512, "photoWidth": 2016, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 480, "videoStabilizationModes": ["auto", "off"], "videoWidth": 640}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb", "p3-d65"], "fieldOfView": 100.46436309814453, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 2}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 94.5, "minISO": 32, "photoHeight": 1512, "photoWidth": 2016, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 480, "videoStabilizationModes": ["auto", "off"], "videoWidth": 640}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb"], "fieldOfView": 107.80342864990234, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 2}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 124.125, "minISO": 32, "photoHeight": 2384, "photoWidth": 4224, "supportsPhotoHDR": false, "supportsVideoHDR": true, "videoHeight": 540, "videoStabilizationModes": ["auto", "cinematic", "off", "standard", "cinematic-extended"], "videoWidth": 960}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb", "p3-d65"], "fieldOfView": 107.80342864990234, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 2}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 124.125, "minISO": 32, "photoHeight": 2384, "photoWidth": 4224, "supportsPhotoHDR": false, "supportsVideoHDR": true, "videoHeight": 540, "videoStabilizationModes": ["auto", "cinematic", "off", "standard", "cinematic-extended"], "videoWidth": 960}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb"], "fieldOfView": 100.46436309814453, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 171.875, "minISO": 32, "photoHeight": 3024, "photoWidth": 4032, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 768, "videoStabilizationModes": ["auto", "off", "standard"], "videoWidth": 1024}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb", "p3-d65"], "fieldOfView": 100.46436309814453, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 171.875, "minISO": 32, "photoHeight": 3024, "photoWidth": 4032, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 768, "videoStabilizationModes": ["auto", "off", "standard"], "videoWidth": 1024}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb"], "fieldOfView": 107.80342864990234, "frameRateRanges": [{"maxFrameRate": 30, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 123.75, "minISO": 32, "photoHeight": 2376, "photoWidth": 4224, "supportsPhotoHDR": false, "supportsVideoHDR": true, "videoHeight": 720, "videoStabilizationModes": ["auto", "cinematic", "off", "standard", "cinematic-extended"], "videoWidth": 1280}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb", "p3-d65"], "fieldOfView": 107.80342864990234, "frameRateRanges": [{"maxFrameRate": 30, "minFrameRate": 1}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 123.75, "minISO": 32, "photoHeight": 2376, "photoWidth": 4224, "supportsPhotoHDR": false, "supportsVideoHDR": true, "videoHeight": 720, "videoStabilizationModes": ["auto", "cinematic", "off", "standard", "cinematic-extended"], "videoWidth": 1280}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb"], "fieldOfView": 107.80342864990234, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 2}], "isHighestPhotoQualitySupported": false, "maxISO": 3072, "maxZoom": 124.125, "minISO": 32, "photoHeight": 2384, "photoWidth": 4224, "supportsPhotoHDR": false, "supportsVideoHDR": true, "videoHeight": 720, "videoStabilizationModes": ["auto", "cinematic", "off", "standard", "cinematic-extended"], "videoWidth": 1280}
 LOG  {"autoFocusSystem": "phase-detection", "colorSpaces": ["srgb", "p3-d65"], "fieldOfView": 107.80342864990234, "frameRateRanges": [{"maxFrameRate": 60, "minFrameRate": 2}], "isHighestPhotoQualitySupported": false, "maxISO": 3072 [... cut ...]
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

dyelax, 2022-03-05:
Answering my own question here — this was a problem because I was getting the devices using `useCameraDevices()` with no argument. If I pass an argument, all of the format options show up. Not 100% sure why this is, but it resolved my issue.

```
const devices = useCameraDevices();
const formats = devices.back.formats;  // >60fps formats unavailable

const devices = useCameraDevices("wide-angle-camera"); // or "ultra-wide-angle-camera"
const formats = devices.back.formats;  // >60fps formats available
```

closed 2022-03-05 by dyelax

mrousavy, 2022-03-07:
Hi, yes that's because a different camera device is used when no argument is passed :)

med1esn, 2023-01-19:
hi, i have the same issue, I can get 60FPS but I can not get more than this. even when i selecet diffrent camera . should i specify that i want to use high FPS mode to show me the different formats?