margelo/react-native-vision-camera #869: 🐛 capture/no-valid-data

Srinivas2445, 2022-03-07:
### What were you trying to do?

Got this error while start the recording, faced this issue from 3 days, before that, its works fine

### Reproduceable Code

```tsx
<Camera
          ref={cameraRef}
          style={styles.camera}
          isActive
          video
          
          audio
          preset="vga-640x480"
          device={
            route.params.deviceData && isBack
              ? route.params.deviceData.back
              : route.params.deviceData.front
          }
        />
```

### What happened instead?

[image]

### Device

pixel 3

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-03-07:
Maybe you called `stopRecording` immediately after `startRecording`?

Srinivas2445, 2022-03-07:
> Maybe you called `stopRecording` immediately after `startRecording`?

Thanks lot..

closed 2022-03-07 by Srinivas2445