margelo/react-native-vision-camera #872: Running vision-camera on react-native-0.68-rc-2❓

mcousillas6, 2022-03-07:
### Question

Not sure if this should go under questions or issues. 

We are working on integrating this library on an app targeting react-native 0.68-rc2 (we need the jdk and fabric support for other stuff, so we are using the rc to prototype things for now) and we are having some build issues on Android with vision-camera, specifically with the `extractJNIFiles` build step.

```
FAILURE: Build failed with an exception.

* Where:
Build file '/---------/node_modules/react-native-vision-camera/android/build.gradle' line: 473

* What went wrong:
Execution failed for task ':react-native-vision-camera:extractJNIFiles'.
> Cannot expand ZIP '/---------/node_modules/react-native-reanimated/android/react-native-reanimated-68-hermes.aar' as it does not exist.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 3m 2s

    at makeError (/---------/node_modules/execa/index.js:174:9)
    at /---------/node_modules/execa/index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (/---------/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:109:5)
    at async Command.handleAction (/---------/node_modules/@react-native-community/cli/build/index.js:192:9)
info Run CLI with --verbose flag for more details.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

As far as I've could investigate, the gradle process on this library tries to find a reanimated `.aar` file that matches the `minor` version from the installed react-native package (found at line 240 of the build.gradle file). 
In the case of `0.68-rc-2` there's no `react-native-reanimated-68-hermes.aar` since reanimated hasn't built it yet as far as I've seen on their repo.

Is there any official way of working around this? 

Thanks!

### What I tried

Right now we got it working by overriding that build.gradle file and hard-coding it to `67` since that's the latest .aar file available.

### VisionCamera Version

2.12.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-03-10:
Hey! Ideally I want to use Reanimated from source (just like on iOS), or as a prefab, for easier integration. Unfortunately the REA team only ships prebuilt .aar files, so for every major RN release we are blocked until REA builds a new version, and ABI safety isn't guaranteed until then.
So there is no official way yet until the REA team decides to ship REA as source (or as a prefab). (cc @piaskowyk ?)

You can patch VisionCamera to find REA 67 aar instead, and hope for ABI stability. Or you can disable Frame Processors, then it builds without all of that :)

mcousillas6, 2022-03-10:
Hey! thanks for the response. 
I ended up going with the git patch approach for now and it's working perfectly! I'm doing OCR so disabling frame processors wasn't an option.

Closing this since it's solved and hopefully people can use the same approach as I did if needed until reanimated ships 68.

closed 2022-03-10 by mcousillas6

mrousavy, 2022-03-10:
Nice! Do you have any feedback for VisionCamera's API and Frame Processor feature?

mcousillas6, 2022-03-10:
> Nice! Do you have any feedback for VisionCamera's API and Frame Processor feature?

So far it's been amazing! The docs are great and the plugin system is awesome (I'm using https://github.com/aarongrider/vision-camera-ocr with 0 issues). 

The only tricky thing was mapping coordinates from the frame to the app's vieport to render overlays. I ended up having to do some calculations since the frame's data is on pixels and on the camera's resolution and the overlays are on the view's coordinate system (I'm using rn-svg to render markers over the text-blocks for example).

Having an example of that on the docs would help a lot, at least something that explains the differences between working with pixels vs points and screen resolutions on mobile 😬. I'm open to contributing to the readme once I have a fully working version of that if you think it's a good addition.

mrousavy, 2022-03-10:
Sure! I think the solution to that wouldn't be docs explaining how to do it (since it's gonna be very very complex) but rather a function that maps camera coordinates to relative or absolute screen coordinates :)

mrousavy, 2022-03-10:
And nice work @aarongrider on the perfectly fine working OCR plugin ;)

piaskowyk, 2022-03-14:
Reanimated is ready to build from the source on the client-side. It will be available in the next release but if you want you can install Reanimated directly from GitHub - now it is possible 😄

VipedL, 2022-05-10:
I am still having problems running react-native-vision-camera with the latest react-native-reanimated on react-native 0.68.2. Should this already be patched. All fine if I disable frameProcessors but sadly that's not an option to my purposes.