margelo/react-native-vision-camera #878: How to get the frame as array or JS/React Image❓

felixdittrich92, 2022-03-07:
### Question

Hi,
i want to work with OpenCV so i need each frames buffer/data how can i do this ? 

Thanks a lot :)

### What I tried

const frameProcessor = useFrameProcessor((frame) => {
    "worklet";
    const width = frame.width;
    const height = frame.height;
    console.log(width, height);
    // TODO: JSI HostObject / ImageHostObject to matrix how ??

   //const srcMat = RNCv.imageToMat(frame.data);
  //const image = new Image(frame.data);

  }, []);

### VisionCamera Version

2.12.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-03-08:
Hi! You need to write native Frame Processor Plugins for that :)

closed 2022-03-08 by mrousavy

kuldip-simform, 2024-11-22:
@felixdittrich92  have you found way for this ?

felixdittrich92, 2024-11-22:
Hi @kuldip-simform :wave:,

Yeah we found a solution (should have an repo invite so you can have a look on it - code could be a bit outdated already)