margelo/react-native-vision-camera #879: Seeing dark screen in Android

ataddesse, 2022-03-07:
### What were you trying to do?

The code I found below is from the getting started guide of this library. I'm getting a dark screen when I'm trying to use it. I have added the necessary permissions in the android manifest file. Am I missing anything else? Is there anyone who's having similar issue?  

### Reproduceable Code

```tsx
function App() {
  const devices = useCameraDevices()
  const device = devices.back

  if (device == null) return <LoadingView />
  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
    />
  )
}
```

### What happened instead?

Seeing dark screen in Android

### Device

Android tablet

### VisionCamera Version

latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-03-10:
What does the Logcat output say?

mrousavy, 2022-03-10:
When reporting an error, "Relevant log output" is important. Don't just leave it out, I can't see your screen so I don't know what's going on when you just tell me that you have a blackscreen.

JulesPatry, 2022-03-10:
This is probably because the simulator does not have any devices available.

if the `device` variable is undefined then you will see a black screen.
```
  const devices = useCameraDevices()
  const device = devices.back
```

ataddesse, 2022-03-11:
Thank you for getting back. 

I got it working now. Even though I added permission for camera in AndroidManifest.xml <uses-permission android:name="android.permission.CAMERA" />, I also had to ask for permission using PermissionsAndroid.PERMISSIONS.CAMERA and useEffect hook.

closed 2022-03-11 by ataddesse

sushantnitk, 2022-03-13:
@ataddesse Hey I am still facing issue with black screen, even after adding permission, Any idea.

ataddesse, 2022-03-13:
Do you also grant the permission from your component file? If not try having a function like this and execute it using a button, inside useEffect or componentDidMount(). 

const requestCameraPermission = async () => {
  try {
    const granted = await PermissionsAndroid.request(
      PermissionsAndroid.PERMISSIONS.CAMERA, 
      {
        title: "Cool Photo App Camera Permission",
        message:
          "Cool Photo App needs access to your camera " +
          "so you can take awesome pictures.",
        buttonNeutral: "Ask Me Later",
        buttonNegative: "Cancel",
        buttonPositive: "OK"
      }
    );
    if (granted === PermissionsAndroid.RESULTS.GRANTED) {
      console.log("You can use the camera");
    } else {
      console.log("Camera permission denied");
    }
  } catch (err) {
    console.warn(err);
  }
};