margelo/react-native-vision-camera #883: 🐛 [iOS] Slow Camera Mount after numerous uses

Cavallando, 2022-03-08:
### What were you trying to do?

Hi there,

Awesome job with this library it has worked great for us so far!

Right now we have an app that allows users to post photos/videos. In our flow, the camera gets mounted and then on capture the media is uploaded. We're seeing that if a user posts a number of photos/videos in a row that the camera component gets slower and slower to mount, taking sometimes around 15-20 seconds to be ready.

The only real change we have made has been upgrading reanimated to 2.4.1 

### What happened instead?

There is no crash and doesnt appear to be an error message. I'd expect that if the camera is unmounted that the next mount would have the same performance as the first mount.

### Device

iPhone 12, iPhone 11 Max, iPhone 11

### VisionCamera Version

2.11.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

PABourdais, 2022-03-09:
Have you tried to look at Memory Use in xcode ?

Cavallando, 2022-03-10:
There weren't any memory issues. We actually confirmed this was an issue with having reanimated 2.4.1 installed im closing this and opening a different one

closed 2022-03-10 by Cavallando