margelo/react-native-vision-camera #888: 🐛 Cannot read property 'id' of undefined

abisalde, 2022-03-10:
### What were you trying to do?

Can read the device camera ID on component mount, I already gone through [Issue #9 ](https://github.com/mrousavy/react-native-vision-camera/issues/9l) but I can seems to get a work around and below it the error from **Android Studio logcat**

### Reproduceable Code

```tsx
import React, {useEffect, useState} from 'react';
import {View, Text, ActivityIndicator} from 'react-native';

import SafeAreaView from 'react-native-safe-area-view';
import {Camera, useCameraDevices} from 'react-native-vision-camera';
// Components
import {AppStatusBar} from '../../../../../components/shared';

// Styles && Colors
import {COLORS, STYLES, FONTS} from '../../../../../constants';
import Styles from './styles';

const ScanPreview = (): JSX.Element => {
  const [hasPermission, setHasPermission] = useState<boolean | any>(false);
  const devices = useCameraDevices();
  const device: any = devices.back;

  const requestCameraPermission = async (): Promise<any> => {
    const status = await Camera.requestCameraPermission();
    setHasPermission(status === 'authorized');
  };

  useEffect(() => {
    requestCameraPermission();
  }, []);

  if (!hasPermission) {
    <View style={[STYLES.container]}>
      <Text style={[FONTS.h3, {color: COLORS.secondary}]}>No Permission</Text>
    </View>;
  }

  if (device === null) return <ActivityIndicator color={COLORS.primary} />;

  return (
    <SafeAreaView>
      <AppStatusBar backgroundColor={COLORS.transparent} />
      <Camera style={Styles.camera} device={device} isActive={true} />
    </SafeAreaView>
  );
};

export default ScanPreview;
```

### What happened instead?

The app crashes with **Cannot read property "id" of undefined

### Relevant log output

```shell
`2022-03-10 03:39:44.319 26267-26267/com.patientui E/GraphicExt: Can't load libboost_ext_fwk
2022-03-10 03:39:44.319 26267-26267/com.patientui E/GraphicExt: GraphicExtModuleLoader::CreateGraphicExtInstance false
2022-03-10 03:39:44.541 26267-26350/com.patientui E/gralloc: Arm Module v1.0
2022-03-10 03:39:44.546 26267-26350/com.patientui E/ion: ioctl c0044901 failed with code -1: Invalid argument
2022-03-10 03:39:44.816 26267-26267/com.patientui E/: [onConstructor]: an error ged handle
2022-03-10 03:39:44.816 26267-26267/com.patientui E/GraphicExt: GraphicExtModuleLoader::CreateGraphicExtInstance false
2022-03-10 03:41:02.519 26267-26461/com.patientui E/ReactNativeJS: TypeError: Cannot read property 'id' of undefined
    
    This error is located at:
        in Camera (at scanPreview/index.tsx:41)
        in RCTView (at View.js:32)
        in View (at createAnimatedComponent.js:211)
        in AnimatedComponent (at createAnimatedComponent.js:264)
        in AnimatedComponentWrapper (at src/index.tsx:85)
        in SafeAreaView (at scanPreview/index.tsx:39)
        in ScanPreview (at App.tsx:29)
        in Provider (at App.tsx:28)
        in RNCSafeAreaProvider (at SafeAreaContext.tsx:76)
        in SafeAreaProvider (at App.tsx:26)
        in App (at renderApplication.js:50)
        in RCTView (at View.js:32)
        in View (at AppContainer.js:92)
        in RCTView (at View.js:32)
        in View (at AppContainer.js:119)
        in AppContainer (at renderApplication.js:43)
        in PatientUI(RootComponent) (at renderApplication.js:60), js engine: hermes`
```

### Device

Nokia 3.1

### VisionCamera Version

"2.12.0"

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-03-10:
Hi! You are performing an explicit null check (`device === null`), but device is `undefined`, not `null`. So you should instead do `device == null` (two `=` signs)

closed 2022-03-10 by mrousavy

abisalde, 2022-03-10:
Thanks @mrousavy  Works now