margelo/react-native-vision-camera #893: ✨ Add record / save path

hirbod, 2022-03-12:
### What feature or enhancement are you suggesting?

I think it would be nice to define where the file should be saved. Currently it's being saved in the /tmp/ folder.
While this is not an issue but more a convenience and speed thing,  (I am using FileSystem.copyAsync from expo to create a copy of it, since I am not allowed to "moveAsync" (permission issue?)

I would love to be able to define a save path, which Expo's FileSystem has access to.

Why I am asking for it? Because it's unreliable when the OS clears the tmp folder. I currently generated 16 GB of video files in my tests and there is no way for me to delete those. I think the end user will eventually delete our app if they can't free up space. 

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

There are no workarounds. I tried to delete the files with `FileSystem.deleteAsync` but it always ends up with "file is not deleteable/moveable".

The copies of the files I create though, which live in FileSystems documentDirectory, are move- and deleteable.

I am using an video editor sdk which also has the ability to define the save path and its working pretty good and reliable. 

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2022-03-14:
Thanks for the feature request, that's a pretty valid enhancement. I'd imagine the user being responsible for creating all of the paths and making sure VisionCamera has permission to create a file there?

hirbod, 2022-03-14:
Here are the comments the other SDK I am using is making:

```
* The filename for the exported data .
     * The correct filename extension will be automatically added
     * based on the selected export format. It can be an absolute path or file URL or a relative path.
     * If some relative path is chosen it will be created in a temporary system directory and overwritten
     * if the corresponding file already exists. If the value is `null` an new temporary file will be
     * created for every export.
     * @note Please make sure that the provided `filename` is valid for the different devices and that
     * your application has the corresponding access rights to write to the desired location. For Android,
     * you will want to make sure to set this inside one of the directories conforming to scoped storage:
     * - DCIM/
     * - Pictures/
     * 
     * For Videos you can additionally use:
     *  - Movies/
```

And I am supplying something like this:
```
filename: VIDEO_TMP_DIR + uuidv4()
```

The SDK adds .mov/mp4 automatically based on filetype/codec.

And my VIDEO_TMP_DIR is 
```
export const VIDEO_TMP_DIR = `${FileSystem.documentDirectory}myCoolApp/video-tmp-exports/`;
```

alexstanbury, 2022-03-21:
+1 on this. I have faced issues in the past with losing pictures after a crash because [React Native iOS clears the tmp upon each col boot](https://github.com/facebook/react-native/blob/99f706e5df45d2390d94de39bf0964e6ab923e91/React/Base/RCTUtils.m#L861). Since then I have always moved every capture to a permanent directory immediately after capture, but this can add a slight delay which I'd rather avoid. Another reason is I'd like to generate my own file names instead of mrousavyXXXXXXXXX.jpg as they show up in certain file browsers, so currently I have to do another file operation to rename the file.

garyhlusko, 2022-03-31:
@mrousavy I'm willing to help out with this, giving your approval and kind of a guide. I need this anyway :)

mrousavy, 2022-04-04:
Sure:

1. Add `path?: string` in this interface: https://github.com/mrousavy/react-native-vision-camera/blob/3850491b9f443257880964a4b0c0c70f484ab4f3/src/VideoFile.ts#L17
2. [iOS]: Read `options.path` and if it exists, use it instead of `tempURL` here: https://github.com/mrousavy/react-native-vision-camera/blob/3850491b9f443257880964a4b0c0c70f484ab4f3/ios/CameraView%2BRecordVideo.swift#L39
3. [Android]: Read `options.path` and if it exists, use it instead of `tempURL` here: https://github.com/mrousavy/react-native-vision-camera/blob/3850491b9f443257880964a4b0c0c70f484ab4f3/android/src/main/java/com/mrousavy/camera/CameraView%2BRecordVideo.kt#L41

Test both with and without `path` to make sure everything still works, then create a PR and I can test :)

mrousavy, 2022-04-04:
We need some kind of error handling as well, so e.g.

* Path does not exist
* Path is not valid
* Path is not writeable
* Path is a folder, not a file
* Path has invalid extension (not `.mp4`)

On Android, add here: https://github.com/mrousavy/react-native-vision-camera/blob/main/android/src/main/java/com/mrousavy/camera/Errors.kt
On iOS, add here: https://github.com/mrousavy/react-native-vision-camera/blob/main/ios/CameraError.swift
For JS, add here: https://github.com/mrousavy/react-native-vision-camera/blob/main/src/CameraError.ts

alexstanbury, 2022-04-20:
I'm happy to work on this next week sometime if @garyhlusko can't.

GreenCurveDev, 2022-08-20:
Hi, did this enhancement ever get done? I am loving the vision camera but have hit a dead end re: using the results as its not possible to even read the photo/video files created using Expo FileSystem. FileSystem.getInfoAsync works but copyASync does not (file "isn't readable"). Any workaround's suggestions greatly appreciated.

mortenmo, 2022-10-05:
I'm having the same problem @GreenCurveDev has. I cannot even do `copyASync` on the file to keep it.

GreenCurveDev, 2022-10-06:
Thanks mortenmo, to work around this I prefixed the file paths with file:// but it definitely felt wrong - I will try your suggestion.

Sid-Turner-Ellis, 2023-01-20:
Is this still in progress?

mrousavy, 2023-01-25:
Yea, this is WIP - need [V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376) for this. If you wanna see this faster, consider sponsoring the V3 efforts :) visioncamera@margelo.io

josh-thompson13, 2023-10-12:
Was this ever completed?

emilje, 2024-07-19:
Stuck with the same issue using expo...

gitn00b1337, 2024-07-22:
Same here, with expo I get:

"Caused by: File '///private/var/mobile/Containers/Data/Application/C998E08F-86F2-4413-BC87-FBB59CAC675E/tmp/9B75D8FA-2216-4E1D-B1A6-822806D3E4DC.jpeg' is not readable"

I'm unable to read the file in any way, other than getting info. Further, I cant use RNFS because it doesnt support expo. I tried prepending the file path with file:// like @GreenCurveDev  mentioned but its not working for me at all. Does anyone have a working example for expo?  

```
const fromPath = // tried `file:///{photo.path}` and file://{photo.path} as well as photo.path. 
await FileSystem.copyAsync({
      from: fromPath,
      to: destinationPath
    })
```

gitn00b1337, 2024-07-22:
For those stuck on expo, the only way I've managed is to use fetch as per the docs. Unfortunately this means no immediate cleanup of the temp file but it'll do

```
const result = await fetch(photo.path);
    const blob = await result.blob();

    const reader = new FileReader();
    reader.readAsDataURL(blob);
    reader.onloadend = async () => {
      const base64data = (reader.result as string).split(',')[1];

      await FileSystem.writeAsStringAsync(destinationPath, base64data, {
        encoding: FileSystem.EncodingType.Base64,
      });
      
      console.log('File saved successfully to:', destinationPath);
      return destinationPath;
    };
```

gwashburne, 2024-07-23:
@mrousavy Is this still WIP? The tmp and expo-file-system don't get along very well. Would this be something a donation page could help make happen?

mrousavy, 2024-07-23:
Nope not WIP - I didn't implement this anymore because V3 (or now V4) was already huge and this was low priority for me since it always just worked for me with tmp files.

But yea I could create a donations page and start building this. I think the implementation plan for me would be to just add a path parameter to takePhoto, takeSnapshot and startRecording - file name itself is still generated randomly, and it will throw if the path is invalid.

mrousavy, 2024-07-23:
Added a polar bade to this issue - $400 is a fair goal for this, I don't think it takes too long.

gwashburne, 2024-07-23:
> Added a polar bade to this issue - $400 is a fair goal for this, I don't think it takes too long. Could be done in less than a week

@mrousavy Sounds good! Perhaps I'm completely missing something, but could you post the link to the donations page? Thanks!

mrousavy, 2024-07-23:
It's the issue right here- in the bottom of the issue description, there's a polar link :)

gwashburne, 2024-07-23:
> It's the issue right here- in the bottom of the issue description, there's a polar link :)

Oh! Wow I completely missed it. Thanks!

mrousavy, 2024-07-23:
did you just fund the entire thing yourself

mrousavy, 2024-07-23:
lmao what a legend - thanks! - I'll get started on implementing that right away by the end of the week! 😂 🚀

gwashburne, 2024-07-23:
Haha well, the company I work for funded it. :)

mrousavy, 2024-07-25:
Started working on it here: https://github.com/mrousavy/react-native-vision-camera/pull/3103 - iOS is done! Android left to do

mrousavy, 2024-07-25:
Android done now as well! A few refactors and two fixes later; we're ready to test - @gwashburne can you test if this works for you?

gwashburne, 2024-07-25:
> Android done now as well! A few refactors and two fixes later; we're ready to test - @gwashburne can you test if this works for you?

This might be user error on my end, but I cloned the repo so I could test locally, and npm install fails:

npm error code ERESOLVE
npm error ERESOLVE unable to resolve dependency tree
npm error
npm error While resolving: react-native-vision-camera@4.5.0
npm error Found: react@18.3.1
npm error node_modules/react
npm error   dev react@"^18.3.1" from the root project
npm error
npm error Could not resolve dependency:
npm error peer react@"18.2.0" from react-native@0.74.2
npm error node_modules/react-native
npm error   dev react-native@"^0.74.2" from the root project
npm error
npm error Fix the upstream dependency conflict, or retry
npm error this command with --force or --legacy-peer-deps
npm error to accept an incorrect (and potentially broken) dependency resolution.

mrousavy, 2024-07-25:
You need to use `yarn`

gwashburne, 2024-07-25:
> You need to use `yarn`

Hmm. Tried to do that but still having issues; can't resolve the path to react-native-vision-camera package on pod install. If it's working for you and others, it's probably an issue on my end switching to yarn from npm. I'm not very familiar with yarn

mrousavy, 2024-07-25:
To run VisionCamera example do this:
```
git clone https://....blablabla
cd react-native-vision-camera
cd package
yarn
cd example
yarn
yarn pods
```

Then open `package/example/ios/VisionCameraExample.xcworkspace` in Xcode, select your phone, and run

gwashburne, 2024-07-26:
Thanks! I followed those steps, but unfortunately, at compile time, got these errors:
[image]

mrousavy, 2024-07-26:
Are you sure you installed pods then? These two files have been added and will only be available after you installed pods. (`cd package/example/ios && pod install` or the `yarn pods` command)

mrousavy, 2024-07-26:
Ah this error is because a `.xcode.env.local` file got generated which probably has a wrong path - I just always delete that file.

But yea sure I'll just try to release a beta for you to test in your app

gwashburne, 2024-07-26:
Aha! Ok, after removing that file, it compiled and installed. Example app seems to be running fine!

mrousavy, 2024-07-29:
Done and released in 4.5.1!! 💪 :) 

Thanks for the sponsoring @gwashburne ❤️

closed 2024-07-29 by mrousavy

SarjuHansaliya, 2024-07-29:
@mortenmo I just tried 4.5.1. what are the possible reason that this error comes
```
{ [capture/file-io-error: An unexpected File IO error occurred! Error: The file “28C67DDD-3528-49DF-9C1B-6F559EDD4836.jpg” doesn’t exist.]
  name: 'capture/file-io-error',
  _code: 'capture/file-io-error',
  _message: 'An unexpected File IO error occurred! Error: The file “28C67DDD-3528-49DF-9C1B-6F559EDD4836.jpg” doesn’t exist.',
  _cause: undefined }
```

SarjuHansaliya, 2024-07-29:
@mrousavy  I just tried 4.5.1. what are the possible reason that this error comes
```
{ [capture/file-io-error: An unexpected File IO error occurred! Error: The file “28C67DDD-3528-49DF-9C1B-6F559EDD4836.jpg” doesn’t exist.]
  name: 'capture/file-io-error',
  _code: 'capture/file-io-error',
  _message: 'An unexpected File IO error occurred! Error: The file “28C67DDD-3528-49DF-9C1B-6F559EDD4836.jpg” doesn’t exist.',
  _cause: undefined }
```

My code is simple
```ts
const data = await camera.current.takePhoto({
  path: `${ReactNativeBlobUtil.fs.dirs.CacheDir}/image.jpeg`,
});
```

SarjuHansaliya, 2024-07-29:
Thanks @mrousavy , yes that is working.

gwashburne, 2024-07-29:
> Done and released in 4.5.1!! 💪 :)
> 
> Thanks for the sponsoring @gwashburne ❤️

You're welcome! Thanks for getting this done so fast and helping me out with testing!

mrousavy, 2024-07-29:
awesome! :)

emilje, 2024-07-31:
Can confirm this works nicely with expo too!

mrousavy, 2024-08-01:
great to hear!

alexstanbury, 2024-08-01:
Thanks so much for getting this done! 

Would it be too much to ask that we are also able to provide the file name ourselves, as a separate option? It could be used in combination with the path parameter or standalone. I need my captured images to follow a certain naming convention, so I can't benefit from the performance/convenience benefit of this change because I still need to touch the file system to do the rename.

mrousavy, 2024-08-01:
Out of curiousity - why would you need your filenames to have a different name? How is that affecting the user of your app?

alexstanbury, 2024-08-01:
I use a timestamp for the filename to be able to quickly run certain analytical tasks on the images captured.

gwashburne, 2024-08-07:
@mrousavy Got a bit of a problem when compiling and running the release version: `[capture/file-io-error: An unexpected File IO error occurred! Error: The file “AF0AED19-A636-495D-A3D6-3F1223D64BDF.jpg” doesn’t exist.`
Which is a bit confusing, since I'm only giving it the directory path, not the file name

emilje, 2024-08-07:
@gwashburne I had a similar error and then realized that the folder I was transferring to did not exist and it would not automatically create one just by specifying the path, so I now check and create the folder if it does not exist before taking a photo.

gwashburne, 2024-08-07:
> @gwashburne I had a similar error and then realized that the folder I was transferring to did not exist and it would not automatically create one just by specifying the path, so I now check and create the folder if it does not exist before taking a photo. Did not yet try in production build...
> 
> 

Thanks @emilje! That was exactly the issue. Got it fixed!

gwashburne, 2024-11-13:
> Thanks so much for getting this done!
>
> Would it be too much to ask that we are also able to provide the file name ourselves, as a separate option? It could be used in combination with the path [...]

@alexstanbury @alexrififi I have created a new fork/branch to add this feature. If you'd like to help out with testing it, I'd much appreciate it! I think it's ready, but was hoping for feedback before I make the PR. It's [available here](https://github.com/gwashburne/react-native-vision-camera/tree/feat/custom-name)