margelo/react-native-vision-camera #897: ❓ Should useCameraDevices always return the "default" device? Front camera in black and white

jladuval, 2022-03-14:
### Question

I'm not sure if this is a bug. I'm using a Pixel 4 and the default front camera appears to be the infrared camera (it's grainy and in black and white). When using the back camera I seem to get the "default" camera. I say default because it's the same one other apps seem to use or other libraries return.

Should useCameraDevices return the "default" front and back cameras? It seems like that' would make the most sense otherwise you'd have to be filtering cameras for each device. I've looked through the other issues and I can't see anyone else hitting this. Is this something I'm doing wrong?

### What I tried

```
const Foo = () => {
   const devices = useCameraDevices()
   const device = devices.front
   return (
       <Camera
          ref={(ref) => (cameraRef.current = ref)}
          device={device}
          isActive={true}
        />
    )
}

```

### VisionCamera Version

2.12.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-03-14:
Lol that's definitely a bug. You could try filtering devices yourself, it's pretty straight forward:

```ts

useEffect(() => {
  const devices = await Camera.getAvailableDevices()
  // ...
}, []
```

jladuval, 2022-03-15:
Ha ok good to know. It's a relatively niche feature for my app so I think Pixel 4 users will just have to deal with their new black and white selfie feature :smile:

dylan-westbury, 2022-05-18:
Also facing this issue, front camera only records in black / white, back camera fine

dylan-westbury, 2022-05-26:
It appears the "wide-angle-camera" is black and white for the front camera on pixels. 

We resolved by filtering for the "ultra-wide-angle-camera"

```
 const wideAngleCamera = frontDevices.filter((dev) =>
          dev.devices.includes("ultra-wide-angle-camera")
        );
```

MorvanL, 2022-11-15:
Black and white photos for us on huawei P20 and huawei P20 Pro. Seems to be with back wide-angle camera

terryatgithub, 2023-01-20:
we met the exactly issue in Google Pixel 4, it seems the issue because the result of Camera.getAvailableCameraDevices() is not consistent. as following screenshot:
1. in react-native-vision-camera/src/hooks/useCameraDevices.ts
the Camera.getAvailableCameraDevices() sorts 'wide-angle-camera' first, so useCameraDevices() would use this one as front camera.
2. while in our app, 
the Camera.getAvailableCameraDevices() sorts 'ultra-wide-angle-camera' first

[image]

[image]

mrousavy, 2023-01-20:
> sorts first

No one said that Camera.getAvailableDevices does do any sorting. You are responsible for sorting that list.

mrousavy, 2023-09-30:
This should be now fixed with the all new Devices API in V3! 🥳

closed 2023-09-30 by mrousavy