margelo/react-native-vision-camera #906: ERROR  TypeError: undefined is not an object (evaluating 'device.id')

zengqinghua1, 2022-03-15:
### What were you trying to do?

import {Camera, useCameraDevices} from 'react-native-vision-camera';

const App = () => {
  const devices = useCameraDevices();
  const device = devices.back;

  return (
    <Camera
      style={StyleSheet.absoluteFill}
      isActive={true}
      photo={true}
      device={device}
    />
  );
};

### What happened instead?

After integrating the plug-in, use the above code to report an error

### Device

xiaomi mi8

### VisionCamera Version

2.12.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

chenshaxuanjing, 2022-03-15:
same issues

mrousavy, 2022-03-15:
The code above **does not** throw that error that you say it does. 

Issue template is not filled out correctly, you did not share the full code, nor did you paste any relevant log output.

Also, I believe you are simply ignoring the fact that `device` can still be null at that point and you need an early return.

closed 2022-03-15 by mrousavy

zengqinghua1, 2022-03-15:
My new project did not do anything else but added the following code in app.js[image]

zengqinghua1, 2022-03-15:
I uploaded a picture. You can have a look

https://user-images.githubusercontent.com/49550422/158347322-ad6eafe4-5f01-4797-9dff-986984011a75.png

mrousavy, 2022-03-15:
Exactly, now I can see your mistake. You did not paste that code in your original issue.

Your mistake is, as I said in my comment before, that you do not check the `device` for null first. It is still loading, you need to return early until it is actually loaded

mrousavy, 2022-03-15:
in VisionCamera's README:

[image]

Lg0gs, 2022-03-30:
@mrousavy 
Didn't help me!!

gagtic, 2022-04-24:
just console logged the useCameraDevices, got this response

`{
   "front":{
      "formats":[
         {
            "pixelFormat":"420v",
            "videoStabilizationModes":[
               "off"
            ],
            "maxZoom":8,
            "minISO":100,
            "maxISO":1000,
            "autoFocusSystem":"none",
            "supportsPhotoHDR":false,
            "colorSpaces":[
               "jpeg"
            ],
            "videoWidth":1920,
            "videoHeight":1440,
            "photoWidth":1920,
            "frameRateRanges":[
               {
                  "maxFrameRate":30,
                  "minFrameRate":1
               },
               {
                  "maxFrameRate":15,
                  "minFrameRate":15
               },
               {
                  "maxFrameRate":30,
                  "minFrameRate":7
               },
               {
                  "maxFrameRate":30,
                  "minFrameRate":15
               },
               {
                  "maxFrameRate":30,
                  "minFrameRate":30
               }
            ],
            "photoHeight":1440,
            "supportsVideoHDR":false,
            "isHighestPhotoQualitySupported":true,
            "fieldOfView":65.59413150556766
         },
         {
            "pixelFormat":"420v",
            "videoStabilizationModes":[
               "off"
            ],
            "maxZoom":8,
            "minISO":100,
            "maxISO":1000,
            "autoFocusSystem":"none",
            "supportsPhotoHDR":false,
            "colorSpaces":[
               "jpeg"
            ],
            "videoWidth":1920,
            "videoHeight":1080,
            "photoWidth":1920,
            "frameRateRanges":[
               {
                  "maxFrameRate":30,
                  "minFrameRate":1
               },
               {
[... 1825 lines cut ...]
            "fieldOfView":65.59413150556766
         }
      ],
      "maxZoom":8,
      "minZoom":1,
      "supportsLowLightBoost":false,
      "supportsDepthCapture":false,
      "neutralZoom":1,
      "supportsFocus":true,
      "supportsRawCapture":false,
      "hasFlash":true,
      "name":"front (1)",
      "supportsParallelVideoProcessing":true,
      "isMultiCam":true,
      "hasTorch":true,
      "devices":[
         "telephoto-camera",
         "wide-angle-camera"
      ],
      "position":"front",
      "id":"1"
   }
}`

There's no back camera available, even if the android device does have it. can someone please confirm if they also face this issue?

mrousavy, 2022-04-25:
Lol, I've never seen this before.

Maybe try using the `Camera.getAvailableCameraDevices()` function and check what that gives you?

Rajan-Savaliya, 2022-05-12:
i am facing the same issue

Psionyx, 2022-06-16:
I was having similar trouble, got it to work by testing for device, but I also had to specify which camera I was using in useCameraDevices.

const devices = useCameraDevices('wide-angle-camera');
const device = devices.back;

Then as @mrousavy mentioned above you have to test to make sure device isn't null

Uzef1997, 2022-08-23:
facing the same issue with     const devices = useCameraDevices() while Camera.getAvailableCameraDevices() is returning detail of my camera can someone help 
Thank you

eldorne, 2022-08-26:
I've got this problem and i solve it by replacing `if (device === null) return <LoadingView />` by `if (device == null) return <LoadingView />` in my code and it worked

Uzef1997, 2022-08-31:
Doesn't matter now in my case I was using react native vision camera for the purpose of barcode scanning but I found an alternative for scanning barcode and QRcode though it is not tested in ios from my side hopefully I would work but if I got any issues then I have to shift back to vision camera, Here is the alternative I found useful 
[react-native-camera-kit](https://www.npmjs.com/package/react-native-camera-kit)

Thank you for the response