margelo/react-native-vision-camera #907: 🐛 App crash for LEGACY camera device with video set to true

capriine, 2022-03-15:
### What were you trying to do?

Using the example app from 2.12.2, the app crashes for me when using video={true} for the Camera, running on an older Android 7.1.1 Motorola E4 (Plus) which has LEGACY support for Camera2 API.

Photos seem to be fine.

Things I have checked or tried:
- I have tried photo={false},  photo={undefined} and omitting photo completely.
- I checked out the supportsParallelVideoProcessing on https://github.com/mrousavy/react-native-vision-camera/blob/c627febde5c51336be058172d0ed4d19ea47d697/docs/docs/guides/DEVICES.mdx#the-supportsparallelvideoprocessing-prop and everything seems to make sense, the value is false, but I am only trying to do one thing - video.
- I have tried disableFrameProcessors in gradle.properties, cleaning and re-building.
- I have tried this on another Motorola phone, running Android 11 and everything seems fine.
- If I omit photo and video (or set them both to be false) I see the preview - so it does seem to be related to the preview and the video being captured simultaneously.
- I have tried video={true} with varying values for preset and for format, none seem to work.
- I tried removing a lot more of the properties for the Camera as well, taking it back to the simple versions in the Guides https://mrousavy.com/react-native-vision-camera/docs/guides/
- I also removed Reanimated and tried Camera on its own.
- I can see changes last year to separate the use cases, but the issue still persists for me, so it must be something else in this specific case.

When using Camera2 Probe, my device shows as LEGACY as expected, but after reading around a lot it seems I should be able to record video or photo independently with that.
[image]

Any advice or suggestions would be very welcome, and if anything else is needed please let me know.
I am happy to help any way I can, as it may help others with some legacy device support.

### Reproduceable Code

```tsx
// Example app from 2.12.2 with one change, photo={false}

              <ReanimatedCamera
                ref={camera}
                style={StyleSheet.absoluteFill}
                device={device}
                format={format}
                fps={fps}
                hdr={enableHdr}
                lowLightBoost={device.supportsLowLightBoost && enableNightMode}
                isActive={isActive}
                onInitialized={onInitialized}
                onError={onError}
                enableZoomGesture={false}
                animatedProps={cameraAnimatedProps}
                photo={false}
                video={true}
                audio={hasMicrophonePermission}
                frameProcessor={device.supportsParallelVideoProcessing ? frameProcessor : undefined}
                orientation="portrait"
                frameProcessorFps={1}
                onFrameProcessorPerformanceSuggestionAvailable={onFrameProcessorSuggestionAvailable}
              />
```

### What happened instead?

App crashes with generic message
[image]

### Relevant log output

```shell
// There is more before this, but github won't let me paste it all in for length reasons, so I can add that if needs be via a comment. I couldn't see anything of interest in that part at a glance.
2022-03-15 10:12:04.226 2847-2917/com.mrousavy.camera.example I/VisionCamera: Initializing FrameProcessorRuntimeManager...
2022-03-15 10:12:04.226 2847-2917/com.mrousavy.camera.example I/VisionCamera: Initializing Vision JS-Runtime...
2022-03-15 10:12:04.233 2847-2917/com.mrousavy.camera.example I/VisionCamera: Initialized Vision JS-Runtime!
2022-03-15 10:12:04.233 2847-2917/com.mrousavy.camera.example I/FrameProcessorRuntime: Installing Frame Processor Plugins...
2022-03-15 10:12:04.234 2847-2917/com.mrousavy.camera.example I/VisionCamera: Installing Frame Processor Plugin "__example_plugin"...
2022-03-15 10:12:04.234 2847-2917/com.mrousavy.camera.example I/FrameProcessorRuntime: Successfully installed 1 Frame Processor Plugins!
2022-03-15 10:12:04.234 2847-2917/com.mrousavy.camera.example I/FrameProcessorRuntime: Installing JSI Bindings on JS Thread...
2022-03-15 10:12:04.419 2847-2914/com.mrousavy.camera.example D/SoLoader: About to load: libreactnativeblob.so
2022-03-15 10:12:04.420 2847-2914/com.mrousavy.camera.example D/SoLoader: libreactnativeblob.so not found on /data/data/com.mrousavy.camera.example/lib-main
2022-03-15 10:12:04.420 2847-2914/com.mrousavy.camera.example D/SoLoader: libreactnativeblob.so found on /data/app/com.mrousavy.camera.example-1/lib/arm
[... 392 lines cut ...]
        at android.os.Handler.dispatchMessage(Handler.java:103)
        at android.os.Looper.loop(Looper.java:203)
        at android.app.ActivityThread.main(ActivityThread.java:6339)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1084)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:945)
2022-03-15 10:12:13.754 2847-2856/com.mrousavy.camera.example D/Camera-JNI: (tid:2856)[release camera] - X context=0x0 
2022-03-15 10:12:13.754 2847-2856/com.mrousavy.camera.example D/Camera-JNI: (tid:2856)[release camera] - X context=0x0 
2022-03-15 10:17:03.513 2847-3221/? I/System.out: close [socket][/127.0.0.1:54587]
```

### Device

Motorola E4 Plus, Android 7.1.1

### VisionCamera Version

2.12.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-03-15:
Hey! 

Thank you so much for the detailed issue report. This is nice to work with.

Have you tried removing `frameProcessor` as well?

capriine, 2022-03-15:
Thanks for the reply, really appreciate it.

And no worries, I wouldn't even raise a ticket if I hadn't been through as much as I could without running out of ideas, so everything I found was put onto the ticket to try and help out.

I just tried the example app again without `frameProcessor`, and the same happens.
The log output to logcat is identical (except from the timestamps)

Also ran it with
```
              <ReanimatedCamera
                ref={camera}
                style={StyleSheet.absoluteFill}
                device={device}
                // format={format}
                preset={'low'}
                // fps={fps}
                // hdr={enableHdr}
                // lowLightBoost={device.supportsLowLightBoost && enableNightMode}
                isActive={isActive}
                // onInitialized={onInitialized}
                // onError={onError}
                // enableZoomGesture={false}
                // animatedProps={cameraAnimatedProps}
                photo={false}
                video={true}
                audio={false}
                // frameProcessor={device.supportsParallelVideoProcessing ? frameProcessor : undefined}
                // orientation="portrait"
                // frameProcessorFps={1}
                // onFrameProcessorPerformanceSuggestionAvailable={onFrameProcessorSuggestionAvailable}
              />```
which was also the same.

huseyintamer, 2022-05-06:
Anyone found a solution?

huseyintamer, 2022-05-08:
I found solution, my device is OPPO X3 Pro - Android 12

```
"react-native": "0.66.3",
"react-native-vision-camera": "^2.13.3",
```

1. Remove package
2. Clean phone cache
3. Restart phone
4. rm -rf node_modules
5. npm install

its fixed

capriine, 2022-05-27:
Just got around to trying this out, and I am still seeing the same behaviour.

I have tried the steps from above by huseyintamer as well as trying a factory reset on the device and a brand new copy of the example project with the latest 2.13.3 release.

ssoruf, 2022-07-11:
News?

Sidebook, 2022-07-18:
Just as another data point, I'm seeing the same issue on Xiaomi Redmi Note 11 Pro+ 5G w/ Android 12.

sc11p, 2022-07-25:
Same on POCO M4 Pro 5G/ Android 12

ckaznable, 2022-07-28:
Based on https://github.com/android/camera-samples/issues/434 I upgraded androidx.camera version to 1.1.0-rc01 and solved the crash problem.

```
implementation "androidx.camera:camera-core:1.1.0-rc01"
implementation "androidx.camera:camera-camera2:1.1.0-rc01"
implementation "androidx.camera:camera-lifecycle:1.1.0-rc01"
implementation "androidx.camera:camera-video:1.1.0-rc01"

implementation "androidx.camera:camera-view:1.1.0-rc01"
implementation "androidx.camera:camera-extensions:1.1.0-rc01"
```

However, I still can't record and the following error is displayed.

```
 "code": "capture/recorder-error", "message": "[capture/recorder-error] The recording failed because the Recorder is in an unrecoverable error state.
When this happens and the output file is generated, the output file is not properly constructed. The application will need to clean up the output file, such as deleting the file. Such an error will usually require creating a new Recorder object to start a new recording.", "userInfo": null}
```

My device is Redmi 10C/ Android 11.

Sidebook, 2022-08-25:
https://github.com/android/camera-samples/issues/450 also seems related, upgrading androidx.camera version to 1.2.0 may solve the issue.
(I haven't been able to find time to try it, though)

adembacajmemmo, 2022-09-23:
Did anyone managed to solve this issue? I just got some reports from Bugsnag from users with Oppo devices. They ahd exactly the same issue.

hirbod, 2022-09-24:
Looks like the issue is solved with the camera-library 1.2.0-beta02 - but this requires to bump the compileSdkVersion to 33, which currently does break `expo-dev-menu` for me. So I could not finally verify it.

hirbod, 2022-09-24:
I got reports back that changing to 

```
  implementation "androidx.camera:camera-core:1.1.0"
  implementation "androidx.camera:camera-camera2:1.1.0"
  implementation "androidx.camera:camera-lifecycle:1.1.0"
  implementation "androidx.camera:camera-video:1.1.0"

  implementation "androidx.camera:camera-view:1.1.0"
  implementation "androidx.camera:camera-extensions:1.1.0"
```

was enough and fixed the issues. I don't have reports about recording issues, but those might still happen on other devices. Ultimately, we need an upgrade to 1.2.0

parthasarathi95, 2022-09-29:
Facing same issue on Realme Narzo 30 Pro 5G, while setting the video={true} app is crashing. Any solutions?

sylvobravo, 2022-10-13:
```
implementation "androidx.camera:camera-core:1.2.0-beta02"
    implementation "androidx.camera:camera-camera2:1.2.0-beta02"
    implementation "androidx.camera:camera-lifecycle:1.2.0-beta02"
    implementation "androidx.camera:camera-video:1.2.0-beta02"
    implementation "androidx.camera:camera-view:1.2.0-beta02"
    implementation "androidx.camera:camera-extensions:1.2.0-beta02"
```
adding the above to your build.gradle should probably fix it

adembacajmemmo, 2022-10-14:
> I got reports back that changing to
>
> ```
>   implementation "androidx.camera:camera-core:1.1.0"
>   implementation "androidx.camera:camera-camera2:1.1.0"
>   implementation [...]

This one fixed the issue for us. We have issues on specific devices like Oppo, Motorola etc...

pierroo, 2022-10-14:
I am sorry if the question sounds stupid, but where are we supposed to add those lines above?
And how do we know which version of androidx camera we are actually using?

wygarner, 2022-10-25:
> I am sorry if the question sounds stupid, but where are we supposed to add those lines above?
And how do we know which version of androidx camera we are actually using?

You can create a patch for react-native-vision-camera using `patch-package` and `postinstall-postinstall`

1. `yarn add patch-package postinstall-postinstall --dev`
2. add `"postinstall": "patch-package"` to `"scripts"` in your `package.json`
3. In `node_modules/react-native-vision-camera/android/build.gradle` find the lines mentioned above, change them to the lines in https://github.com/mrousavy/react-native-vision-camera/issues/907#issuecomment-1278290958, then save your changes
4. run `yarn patch-package react-native-vision-camera`

Your patch will be saved in your project directory and will be applied when you run the project

subhadeepquantiantech, 2022-12-21:
> I got reports back that changing to
>
> ```
>   implementation "androidx.camera:camera-core:1.1.0"
>   implementation "androidx.camera:camera-camera2:1.1.0"
>   implementation [...]

working:D

xeroxoid, 2023-01-24:
Can confirm that using version `1.1.0` in `android/app/build.gradle` solves the issue.

xeroxoid, 2023-01-24:
Created PR #1425 which should fix the issue, although I guess it could be updated to 1.2.0 too.

subhadeepquantiantech, 2023-01-24:
implementation "androidx.camera:camera-core:1.1.0"
  implementation "androidx.camera:camera-camera2:1.1.0"
  implementation "androidx.camera:camera-lifecycle:1.1.0"
  implementation "androidx.camera:camera-video:1.1.0"

  implementation "androidx.camera:camera-view:1.1.0"
  implementation "androidx.camera:camera-extensions:1.1.0"

izuutech, 2023-05-13:
> I got reports back that changing to
>
> ```
>   implementation "androidx.camera:camera-core:1.1.0"
>   implementation "androidx.camera:camera-camera2:1.1.0"
>   implementation [...]

This seemed to solve the issue on a Tecno Camon 20 Pro (Android 13) But what exactly does this do

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy