margelo/react-native-vision-camera #908: Error when building for Android 🐛

Altequer, 2022-03-15:
### What were you trying to do?

Whenever I go to build, an error occurs saying that the file is missing. I've tested with other versions but still. Could anyone help?

### Reproduceable Code

```tsx
gradlew assembleRelease
```

### What happened instead?

[image]
[dependencies.txt](https://github.com/mrousavy/react-native-vision-camera/files/8255316/dependencies.txt)

### Relevant log output

```shell
The following error always occurs:
C/C++: ../../../../build/third-party-ndk/folly\folly/dynamic.h:63:10: fatal error: 'boost/operators.hpp' file not found
```

### Device

Android

### VisionCamera Version

2.12.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Jamal-Nazzal, 2022-03-16:
I'm facing the same issue
```
  ../../../../build/third-party-ndk/folly\folly/dynamic.h:63:10: fatal error: 'boost/operators.hpp' file not found
  #include <boost/operators.hpp>
           ^~~~~~~~~~~~~~~~~~~~~
```

mrousavy, 2022-03-16:
Hey!

Could you try running ./gradlew clean before the release build, see if that helps? Does it work in debug?

Jamal-Nazzal, 2022-03-16:
When I try this:
go to \node_modules\react-native-vision-camera\android\build\third-party-ndk
I find the boost folder named "boost_1_63_0", I rename it to just "boost". the build works, but this time I get the following error in the react-native start terminal
```

throw ex;
          ^

Error: EPERM: operation not permitted, lstat '[project directory]\node_modules\react-native-vision-camera\android\build\intermediates\cxx\Debug\4x231b5p\obj\arm64-v8a\libVisionCamera.so.tmp2ab35ca'
Emitted 'error' event on NodeWatcher instance at:
    at NodeWatcher.<anonymous> ([project directory]\node_modules\sane\src\node_watcher.js:291:16)
    at FSReqCallback.oncomplete (node:fs:198:21) {
  errno: -4048,
  code: 'EPERM',
  syscall: 'lstat',
  path: '[project directory]\\node_modules\\react-native-vision-camera\\android\\build\\intermediates\\cxx\\Debug\\4x231b5p\\obj\\arm64-v8a\\libVisionCamera.so.tmp2ab35ca'
}
```

and if I delete the build and .cxx folders, then start clean cache, it builds fine but again crashes if I rebuild.

Altequer, 2022-03-16:
I ran it but the error remains when I generate the apk

[image]

Jamal-Nazzal, 2022-03-16:
Please check my comment above. inside `\node_modules\react-native-vision-camera\android\build\third-party-ndk` there is a folder that should be named `boost` for some reason it gets named `boost_1_63_0` that's why boost/operators.hpp can't be found. 
If you rename it to the correct name `boost` the error goes away. that's not a solution AT ALL, I'm posting this to make you aware of the source of the problem.

Altequer, 2022-03-18:
If removing '_1_63_0' from the name really works, is there any way to definitively resolve this?

Jamal-Nazzal, 2022-03-18:
the first time visionCamera builds it installs this library [https://github.com/react-native-community/boost-for-react-native](url) something wrong happens there. sometimes it builds fine and creates a folder called `boost`, other times it names it `boost_1_63_0` and as a result boost/operators.hpp can't be found. Can't be clearer than this honestly. can this be `definitively` resolved? I don't know, mrousavy might be able to solve it.

jslok, 2022-03-21:
I'm getting this issue when building for production. In development, it builds fine.

RafaelA977, 2022-04-25:
Same issue for me. Vision camera download "boost_${BOOST_VERSION}.tar.gz" from github, unzip it.

After that, in doLast block of the prepareBoost task renames the file to boost:
[image]

but the file is not renamed for some reason. So we got "'boost/operators.hpp' file not found" error.

Manually rename folder after first build works for me

formifan2002, 2022-05-02:
@RafaelA977 found the porblem in the build.gradle. The reason is, that gradle cannot rename the directory. Comment the line

//file("$thirdPartyNdkDir/boost_${BOOST_VERSION}").renameTo("$thirdPartyNdkDir/boost")

and replace it with 

ant.move(file: "$thirdPartyNdkDir/boost_${BOOST_VERSION}", tofile: "$thirdPartyNdkDir/boost", quiet: "false",performGCOnFailedDelete: "true")

Jamal-Nazzal, 2022-05-02:
Now I'm getting this error
`Error: EPERM: operation not permitted, lstat 'C:\Users\...\app\node_modules\react-native-vision-camera\android\build\intermediates\cxx\Debug\602l1f37\obj\arm64-v8a\libVisionCamera.so.tmp50c975e'
Emitted 'error' event on NodeWatcher instance at:
    at NodeWatcher.<anonymous> (C:\Users\...\app\node_modules\sane\src\node_watcher.js:291:16)
    at FSReqCallback.oncomplete (node:fs:198:21) {
  errno: -4048,
  code: 'EPERM',
  syscall: 'lstat',
  path: 'C:\\Users\\...\\app\\node_modules\\react-native-vision-camera\\android\\build\\intermediates\\cxx\\Debug\\602l1f37\\obj\\arm64-v8a\\libVisionCamera.so.tmp50c975e'
}`

MorvanL, 2022-05-04:
Hello ! I have a similar error with folly but i think that is the same : `node_modules/react-native/ReactCommon/cxxreact/CxxModule.h:15:10: fatal error: 'folly/dynamic.h' file not found`

I need to disable FrameProcessors to build (https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processors/#android) but we will soon need it :/

formifan2002, 2022-05-05:
@MorvanL: before first correct "gradlew build" you must (!) run a "gradlew clean". Otherwise the gradlew build might not process correctly since some parts (e.g. folly...tar.gz) have been downloaded already during earlier builds and will not be processed correctly. Check after the clean and before the build, if  directory "react-native-vision-camera\android\build\" is empty. If not, delete any content in the directory manually and start the build.

MorvanL, 2022-05-05:
@formifan2002 I did some tests:
If I only run `gradlew assembleRelease` it works. 

When I run `gradlew clean assembleRelease` I have : success, failed, success, failed, ....
But if I run `gradlew clean` then  `gradlew assembleRelease` it works. 
However in both cases I could see that the "react-native-vision-camera\android\build" file is correctly deleted

On our CI using fastlane and running `gradlew clean assembleRelease`, it fails every time. May be because there is more clean and yarn install between each build

MorvanL, 2022-05-05:
I think we can work around but there must be a problem in the build process

MorvanL, 2022-05-05:
Sorry for spam. 
When it fails I don't have the lib folly in the `react-native-vision-camera/android/build/third-party-ndk` folder
I have only `hermes` and `jsc` folder.

RafaelA977, 2022-05-06:
I found a solution to solve this problem. With this patch for build.gradle it works for me:
[image]

ThaJay, 2022-05-30:
Getting the following error with version 2.13.3:

```
Error: ENOENT: no such file or directory, lstat '/Users/dj/dev/chipta/scan-app/node_modules/react-native-vision-camera/android/build/third-party-ndk/boost_1_63_0
Emitted 'error' event on FSEventsWatcher instance at:
    at Walker.emit (node:events:526:28)
    at /Users/dj/dev/chipta/scan-app/node_modules/walker/lib/walker.js:52:12
    at FSReqCallback.oncomplete (node:fs:198:21) {
  errno: -2,
  code: 'ENOENT',
  syscall: 'lstat',
  path: '/Users/dj/dev/chipta/scan-app/node_modules/react-native-vision-camera/android/build/third-party-ndk/boost_1_63_0'
}
```

ThaJay, 2022-05-31:
If I rename the first folder `boost` to `boost_1_63_0`, I get the following error:
```
In file included from /Users/dj/dev/chipta/scan-app/node_modules/react-native/ReactAndroid/src/main/jni/react/jni/NativeMap.h:11:
  ../../../../build/third-party-ndk/folly/folly/dynamic.h:63:10: fatal error: 'boost/operators.hpp' file not found
  #include <boost/operators.hpp>
           ^~~~~~~~~~~~~~~~~~~~~
```

ThaJay, 2022-06-02:
@RafaelA977 Was it too much trouble to copy the text?

    eachFile { fname -> fname.path = (fname.path - "boost_${BOOST_VERSION}/")}
    into("$thirdPartyNdkDir/boost")

After applying your patch, I get enother error:
```

> Task :app:bundleReleaseJsAndAssets FAILED
/Users/dj/dev/chipta/scan-app/node_modules/metro-hermes-compiler/src/emhermesc.js:77
          throw ex;
          ^

Error: ENOENT: no such file or directory, lstat '/Users/dj/dev/chipta/scan-app/node_modules/react-native-vision-camera/android/.cxx/cmake/release/armeabi-v7a/CMakeFiles/CMakeTmp/CMakeFiles/TargetDirectories.txt'
Emitted 'error' event on FSEventsWatcher instance at:
    at Walker.emit (node:events:526:28)
    at /Users/dj/dev/chipta/scan-app/node_modules/walker/lib/walker.js:52:12
    at FSReqCallback.oncomplete (node:fs:198:21) {
  errno: -2,
  code: 'ENOENT',
  syscall: 'lstat',
  path: '/Users/dj/dev/chipta/scan-app/node_modules/react-native-vision-camera/android/.cxx/cmake/release/armeabi-v7a/CMakeFiles/CMakeTmp/CMakeFiles/TargetDirectories.txt'
}
```

ThaJay, 2022-06-02:
I went ahead and tried older versions. 2.13.1 is the latest version that builds for me.
```"react-native-vision-camera": "2.13.1",```

ThaJay, 2022-06-28:
I don't know why but today I got the error on `2.13.1` but on the latest version (`2.13.5`) it builds production just fine.

ThaJay, 2022-07-07:
For some reason it did not build any more but weirly enough the latest version (2.13.5) did work this time. This confuses me but I'm glad it works.

ApWin77, 2022-08-31:
Hi, @MorvanL . Have you resolved issue in your case ?

MorvanL, 2022-09-09:
Hi, sorry I just saw your message. 
Until now we didn't need the frame processor so we disabled them. But now we need them for our barcode scanner and according to my first tests the problem is still there. It seems that doing `gradlew clean` then `gradlew assembleRelease `does not solve the problem anymore. I haven't tested the patch described above, but is there any chance that the problem will be fixed? 
We use RN 0.69.4 and I tried with and without Hermes enable

MorvanL, 2022-09-09:
Oh my bad, seems working with `gradlew clean` before `gradlew assembleRelease` (and not `gradlew clean assembleRelease` together). 
But I have to test with our CI and for the moment I have another error. I'll let you know once I've tested everything

closed 2023-08-18 by mrousavy