margelo/react-native-vision-camera #909: ✨ Build apk for each release

Angelk90, 2022-03-15:
### What feature or enhancement are you suggesting?

Whenever a release is released, it would be nice for a sample apk to be generated so that you can get a general idea of the sample project.

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

Use the github actions, which every time a release is released, a sample apk is generated.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2022-03-16:
Yeah actually not a bad idea - release builds take a little bit longer than debug but we can hook it to Release Trigger

mrousavy, 2023-09-30:
I don't think this makes sense atm

closed 2023-09-30 by mrousavy