margelo/react-native-vision-camera #914: 🐛 Errors when building for simulator, works fine on devices

ademirmochi, 2022-03-18:
### What were you trying to do?

I'm trying to build my react-native application, it works on devices, but when I try to build on the simulator, it gives me plenty of errors, I've already tried the solutions in the troubleshooting guide, didn't change much.

### What happened instead?

When selecting "any iOS Device arm64" it builds without issues, and I can create the IPA file and test on TestFlight, but when trying to test on simulator, regardless of version, it gives me a collection of erros, usually around the MLKIT components and:

Showing Recent Messages
Undefined symbol: _GULLogBasic
Undefined symbol: _OBJC_CLASS_$_GPBEnumArray
Undefined symbol: _GPBClearOneof
Undefined symbol: _OBJC_CLASS_$_GTMLogMininumLevelFilter
Undefined symbol: _OBJC_METACLASS_$_GTMLogger
Undefined symbol: _OBJC_CLASS_$_GPBCodedOutputStream
Undefined symbol: _OBJC_CLASS_$_GPBFileDescriptor
Undefined symbol: _OBJC_CLASS_$_GPBEnumDescriptor
Undefined symbol: _OBJC_CLASS_$_GPBDescriptor
Undefined symbol: _kGTMSessionFetcherStatusDomain
Undefined symbol: _OBJC_CLASS_$_GDTCORTransport
Undefined symbol: _OBJC_CLASS_$_GTMSessionFetcherService
Undefined symbol: _OBJC_CLASS_$_GTMSessionFetcher
Undefined symbol: _OBJC_METACLASS_$_GPBMessage
Undefined symbol: _OBJC_CLASS_$_GTMLogger
Undefined symbol: _OBJC_CLASS_$_FlipperClient
Undefined symbol: _OBJC_METACLASS_$_GPBRootObject
Undefined symbol: _OBJC_CLASS_$_GULCCComponentContainer
Undefined symbol: _OBJC_CLASS_$_GULCCComponentType
Undefined symbol: _OBJC_CLASS_$_GPBMessage
Undefined symbol: _OBJC_CLASS_$_GULUserDefaults
Undefined symbol: _OBJC_CLASS_$_FlipperKitNetworkPlugin
Undefined symbol: _OBJC_CLASS_$_GPBRootObject
Undefined symbol: _OBJC_CLASS_$_RCTBundleURLProvider
Undefined symbol: _OBJC_CLASS_$_SKIOSNetworkAdapter
Undefined symbol: _OBJC_CLASS_$_FlipperKitReactPlugin
Undefined symbol: _OBJC_CLASS_$_SKDescriptorMapper
Undefined symbol: _OBJC_CLASS_$_GPBInt32Array
Undefined symbol: _OBJC_CLASS_$_GTMSessionCookieStorage
Undefined symbol: _OBJC_CLASS_$_FlipperKitLayoutPlugin
Undefined symbol: _OBJC_CLASS_$_FKUserDefaultsPlugin
Undefined symbol: _OBJC_CLASS_$_GULCCComponent
Undefined symbol: _OBJC_CLASS_$_RCTRootView
Undefined symbol: _OBJC_CLASS_$_RCTBridge

### Relevant log output

```shell
Undefined symbols for architecture x86_64:
  "_GULLogBasic", referenced from:
      _MLKLog in MLKitCommon(MLKLogger.o)
  "_OBJC_CLASS_$_GPBEnumArray", referenced from:
      objc-class-ref in MLKitVision(GMVLogger.o)
  "_GPBClearOneof", referenced from:
      _MLKITx_EPTFlagValue_ClearValueOneOfCase in MLKitCommon(FlagValues.pbobjc.o)
      _MLKITx_PHENORegistrationInfo_ClearPackageVersionOneOfCase in MLKitCommon(RegistrationInfo.pbobjc.o)
      _MLKITx_PHENORegistrationInfo_ClearApplicationInfoOneOfCase in MLKitCommon(RegistrationInfo.pbobjc.o)
      _MLKITx_CLCCCounter_ClearCounterIdOneOfCase in MLKitCommon(Counters.pbobjc.o)
  "_OBJC_CLASS_$_GTMLogMininumLevelFilter", referenced from:
      objc-class-ref in MLKitCommon(GIPLoggingReroutingGTMLogger.o)
  "_OBJC_METACLASS_$_GTMLogger", referenced from:
      _OBJC_METACLASS_$_MLKITx_GIPLoggingReroutingGTMLogger in MLKitCommon(GIPLoggingReroutingGTMLogger.o)
  "_OBJC_CLASS_$_GPBCodedOutputStream", referenced from:
      objc-class-ref in MLKitCommon(CCTLogOutputStream.o)
  "_OBJC_CLASS_$_GPBFileDescriptor", referenced from:
      objc-class-ref in MLKitCommon(Enums.pbobjc.o)
      objc-class-ref in MLKitCommon(UploaderState.pbobjc.o)
      objc-class-ref in MLKitCommon(FirebaseMlSdk.pbobjc.o)
      objc-class-ref in MLKitCommon(Barhopper.pbobjc.o)
      objc-class-ref in MLKitCommon(FlatFile.pbobjc.o)
      objc-class-ref in MLKitCommon(TypedFeatures.pbobjc.o)
[... 113 lines cut ...]
  "_OBJC_CLASS_$_GULCCComponent", referenced from:
      objc-class-ref in MLKitCommon(MLKCustomModelManager.o)
      objc-class-ref in MLKitImageLabeling
  "_OBJC_CLASS_$_RCTRootView", referenced from:
      objc-class-ref in AppDelegate.o
  "_OBJC_CLASS_$_RCTBridge", referenced from:
      objc-class-ref in AppDelegate.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

### Device

iPhone 11 (15) Xcode Simulator

### VisionCamera Version

2.12.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-03-28:
Hey, you did not share any reproduceable code, I cannot reproduce your error.

closed 2022-03-28 by mrousavy

vlorian-de, 2022-04-07:
This issue is already closed. But I had exactly the same problem and found a solution for it.
So maybe it will help somebody in the future, if I share it here:

The build failed only in projects when using the frame processor "react-native-vision-camera".

When I experienced this issue I developed on an MBP with M1Max Apple Silicon running macOS Monterey and XCode 13.3.
I wanted to give it another try with a different macOS / XCode version and did a checkout on my old 2017 Intel based MBP: And everything worked fine. Then I updated my old Intel based MBP to the same versions of node / macOS / XCode and the builds still ran sucessfully.

It seems that the only issue is the hardware platform. Even when working with Rosetta.

**So here is a fix that worked on my Apple silicon MPB:
Make sure that you add "arm64" as "Excluded Architecture" to your project AND the pod configuration:**

[image]

mrousavy, 2022-04-07:
Nice catch @vlorian-de, this shouldn't be happening but glad it's working for you now

ademirmochi, 2022-04-07:
So I'm aware already that the problem was the new chip and all the headache that comes with it, this solution, where you put the "arm64" at excluded architectures, solves a lot of different issues i've been having, in a lot of different libs, but, in this particular case, it didn't

karlmarxlopez, 2022-05-10:
@mrousavy This is still happening on `2.13.3` using the frame processor.

moygospadin, 2022-05-13:
> @ademirmochi Can you confirm that you are also on some kind of M1 chip? Would you like to checkout if you can build my quick and dirty test app? I uploaded it here: https://github.com/vlorian-de/vision-camera-code-scanner-test-app

dont work on MacBook Pro (13-inch, M1, 2020) Xcode Version 13.3 (13E113)

moygospadin, 2022-05-14:
> 
Also happening here

moygospadin, 2022-05-14:
> @ademirmochi Can you confirm that you are also on some kind of M1 chip? Would you like to checkout if you can build my quick and dirty test app? I uploaded it here: https://github.com/vlorian-de/vision-camera-code-scanner-test-app

Also happening here

hdvinhhoang, 2022-05-15:
> This issue is already closed. But I had exactly the same problem and found a solution for it. So maybe it will help somebody in the future, if I share it here:
>
> The build failed only in projects [...]
Thank @vlorian-de. it works for me on M1 chip

moygospadin, 2022-05-16:
@hdvinhhoang could you show your podfile ?

vlorian-de, 2022-05-20:
Hello all,
since people with the same old problem are still coming in here regularly, I wanted to point out the following:

With the current version of ReactNative (0.68.2), the problem no longer occurs (at least for me).

Just try a `npx react-native init YouNameIt`
If that works for you, just update your project to the current dependencies or use a diff to get to the actual error location.

Tested on a MBP2021 M1-Max:
MacOS 12.3.1 (21E258)
XCode 13.3.1 (13E500a)
CommandLine Tools: 13.4.0.0.1.1651278267
ReactNative: 0.68.2

Ace090, 2022-07-29:
> This issue is already closed. But I had exactly the same problem and found a solution for it. So maybe it will help somebody in the future, if I share it here:
>
> The build failed only in projects [...]

It works for me. Thanks!

liptonzuma, 2022-08-19:
> This issue is already closed. But I had exactly the same problem and found a solution for it. So maybe it will help somebody in the future, if I share it here:
>
> The build failed only in projects [...]

So I did this and the problem went away but I deleted the project and cloned the repo for some reason the problem is back with all this configurations in place. Please what do I do?

ospfranco, 2022-11-29:
This is happening to me to:

macbook air m2
xcode 14.1
RN 0.69.7

The workaround on my podfile is to exclude arm64 compilation:

```ruby
  post_install do |installer|
    react_native_post_install(installer)
    __apply_Xcode_12_5_M1_post_install_workaround(installer)
    
    installer.pods_project.targets.each do |target|
      # Prevents dependencies to be compiled under m1
      installer.pods_project.targets.each do |target|
        target.build_configurations.each do |config|
          config.build_settings['EXCLUDED_ARCHS[sdk=iphonesimulator*]'] = "arm64"
        end
      end

      if target.respond_to?(:product_type) and target.product_type == "com.apple.product-type.bundle"
        target.build_configurations.each do |config|
            config.build_settings['CODE_SIGNING_ALLOWED'] = 'NO'
        end
      end
    end
  end
```

comur, 2022-12-09:
Hi all,

This still happens:

macbook air m1
xcode 14.1
RN 0.68.0

Your workaround works @ospfranco ! thanks

Jeppeer, 2023-10-22:
I had the same problem. Upgrading to v3 and removing `vision-camera-code-scanner` solved it.

bumios, 2024-06-29:
> ```
> post_install do |installer|
>   installer.pods_project.targets.each do |target|
>     target.build_configurations.each do |config|
>       config.build_settings['EXCLUDED_ARCHS[sdk=iphonesimulator*]'] = "arm64"
>     end
>   end
> end
> ```

I'm working on MLKit vision only for iOS, but this solution worked for me too, thank you!

hoscahmet, 2024-09-15:
> arm64

> It works for me. Thanks!

nathangabriel27, 2024-09-17:
> This issue is already closed. But I had exactly the same problem and found a solution for it. So maybe it will help somebody in the future, if I share it here:
>
> The build failed only in projects [...]

It works for me. Thanks!