margelo/react-native-vision-camera #919: ❓ fatal error: 'AnimatedSensorModule.h' file not found

rajaryan1990, 2022-03-21:
### Question

In file included from ../../../../src/main/cpp/VisionCamera.cpp:3:
In file included from ../../../../src/main/cpp/FrameProcessorRuntimeManager.h:13:
In file included from /Users/vn537kp/Documents/Workspace/SNG/Mobile_ScanAndGo_RN/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems/RuntimeManager.h:9:
/Users/vn537kp/Documents/Workspace/SNG/Mobile_ScanAndGo_RN/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems/ShareableValue.h:9:10: fatal error: 'AnimatedSensorModule.h' file not found
#include "AnimatedSensorModule.h"
^~~~~~~~~~~~~~~~~~~~~~~~

### What I tried

React Native Version: 0.64.2
Reanimated Version: 2.5.0
Vision Camera version: 2.9.4

Build is running on IOS but failing on Android

### VisionCamera Version

2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

closed 2022-03-21 by mrousavy

rajaryan1990, 2022-03-22:
But If I remove this vision camera library the error is not showing. I am using this reanimated library from long time back but never seen this type of error.

mrousavy, 2022-03-22:
I want to apologize, you're right. I added the missing header and will ship an update soon

DeVoresyah, 2022-04-07:
> I want to apologize, you're right. I added the missing header and will ship an update soon

is this already solved? I had the same issue when using react-native-vision-camera `2.10.0`. And if I use the latest version `2.13.1`, I got an issue from #716

GSFZamai, 2022-04-13:
> > I want to apologize, you're right. I added the missing header and will ship an update soon
> 
> is this already solved? I had the same issue when using react-native-vision-camera `2.10.0`. And if I use the latest version `2.13.1`, I got an issue from #716

Same problem here. Having different errors on each versions. 
Version 2.10.0 don't build on android and version 2.13.1 don't build on iOS.

tatiana-lonestone, 2022-09-05:
Same here, it doesn't build on ios with 2.14.1