margelo/react-native-vision-camera #920: ✨ MacOS catalyst support

Flam3rboy, 2022-03-21:
### What feature or enhancement are you suggesting?

With Catalyst, you can easily compile your existing React-native iOS apps for macOS.
However, it fails in many stages, all with the error "...is only available in Mac Catalyst 14.0 or newer".

### What Platforms whould this feature/enhancement affect?

iOS

### Alternatives/Workarounds

I am currently adding ``Add 'if #available' version `` checks to nearly line, but this isn't feasible for development.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2023-09-30:
This is now implemented with V3! 🥳

closed 2023-09-30 by mrousavy

tommycarpi, 2024-02-03:
I'm still experiencing this in v3.8.2.
Is there any workaround? @mrousavy 

[image]

[image]

Hunnainwali, 2025-04-04:
Facing same issue, my app was crashing on Mac when camera was being rendered multiple times. Tried enabling Mac Catalyst into my project and example project as well but no luck : (