margelo/react-native-vision-camera #921: 🐛 Build failed with real 2.5.0

alexco2, 2022-03-21:
### What were you trying to do?

I wanted to build my app with the newest era v 2.5.0

### Reproduceable Code

```tsx
This bug should be really straightforward to reproduce. If not, I will try to provide a reproducible code asap.
```

### What happened instead?

Build failed with error message

### Relevant log output

```shell
error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081
Note: /Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native-code-push/android/app/src/main/java/com/microsoft/codepush/react/CodePushNativeModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native-code-push/android/app/src/main/java/com/microsoft/codepush/react/CodePushNativeModule.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/alex/Documents/GitHub/TheRightChoice/node_modules/@react-native-firebase/analytics/android/src/main/java/io/invertase/firebase/analytics/UniversalFirebaseAnalyticsModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/alex/Documents/GitHub/TheRightChoice/node_modules/@react-native-firebase/auth/android/src/main/java/io/invertase/firebase/auth/ReactNativeFirebaseAuthModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/alex/Documents/GitHub/TheRightChoice/node_modules/@react-native-firebase/crashlytics/android/src/main/java/io/invertase/firebase/crashlytics/ReactNativeFirebaseCrashlyticsModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/alex/Documents/GitHub/TheRightChoice/node_modules/@react-native-firebase/dynamic-links/android/src/main/java/io/invertase/firebase/dynamiclinks/ReactNativeFirebaseDynamicLinksModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: /Users/alex/Documents/GitHub/TheRightChoice/node_modules/@react-native-firebase/perf/android/src/main/java/io/invertase/firebase/perf/UniversalFirebasePerfModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: /Users/alex/Documents/GitHub/TheRightChoice/node_modules/@react-native-google-signin/google-signin/android/src/main/java/com/reactnativegooglesignin/RNGoogleSigninModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/alex/Documents/GitHub/TheRightChoice/node_modules/@react-native-google-signin/google-signin/android/src/main/java/com/reactnativegooglesignin/RNGoogleSigninModule.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: /Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native-svg/android/src/main/java/com/horcrux/svg/VirtualView.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/alex/Documents/GitHub/TheRightChoice/android/app/src/main/java/com/therightchoice/MainApplication.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:externalNativeBuildDebug'.
> Build command failed.
  Error while executing process /Users/alex/Library/Android/sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a VisionCamera}
  ninja: Entering directory `/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a'
  [1/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o
  [2/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o
  [3/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JImageProxy.cpp.o
  [4/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o
  [5/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o 
  /Users/alex/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=armv7-none-linux-androideabi21 --gcc-toolchain=/Users/alex/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/alex/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/sysroot  -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native/React -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native/React/Base -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native/ReactAndroid/src/main/jni -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native/ReactCommon -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native/ReactCommon/callinvoker -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native/ReactCommon/jsi -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/hermes-engine/android/include -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native-reanimated/Common/cpp/headers/Tools -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native-reanimated/Common/cpp/headers/SpecTools -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native-reanimated/Common/cpp/headers/Registries -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native-reanimated/Common/cpp/headers/LayoutAnimations -I/Users/alex/Documents/GitHub/TheRightChoice/node_modules/react-native-reanimated/Common/cpp/hidden_headers -I../../../../src/main/cpp -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_HAVE_MEMRCHR=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DON_ANDROID -DONANDROID -DFOR_HERMES=true -O0 -fno-limit-debug-info  -fPIC   -std=gnu++14 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -c ../../../../src/main/cpp/VisionCamera.cpp
[... 28 lines cut ...]
BUILD FAILED in 1m 52s

    at makeError (/Users/alex/Documents/GitHub/TheRightChoice/node_modules/execa/index.js:174:9)
    at /Users/alex/Documents/GitHub/TheRightChoice/node_modules/execa/index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (/Users/alex/Documents/GitHub/TheRightChoice/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:109:5)
    at async Command.handleAction (/Users/alex/Documents/GitHub/TheRightChoice/node_modules/@react-native-community/cli/build/index.js:192:9)
[... cut ...]
Samsung Galaxy S21

### VisionCamera Version

^2.11.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

balmasich, 2022-03-22:
Need add 
react-native-vision-camera/android/CMakeLists.txt
after 68 line
"${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/headers/AnimatedSensor"
but work only for new reanimated

mrousavy, 2022-03-22:
Hi, thanks for the report - I fixed this and will ship an update soon

closed 2022-03-22 by mrousavy

francois-spectre, 2022-03-23:
@balmasich Thank you so much, it worked for me 🙏