margelo/react-native-vision-camera #930: 🐛 [ios] Sentry crash report

hirbod, 2022-03-24:
### What were you trying to do?

My user was recording multiple videos. So a bunch of startRecording / stopRecording calls.
From what I can tell, this issue is related to older iPhone devices. My user is using an iPhone SE 2nd Generation with iOS 15.4

The App crashes, heats up a lot and also screen dimming is happening.

### What happened instead?

The app crashes with `Cannot call method when status is 1`

[image]

### Relevant log output

```shell
NSInternalInconsistencyException
*** -[AVAssetWriter finishWritingWithCompletionHandler:] Cannot call method when status is 1
```

### Device

iPhone SE 2nd Generation (15.4)

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-03-28:
Not sure what status 1 is, I'd have to read AVFoundation headers again. Maybe not enough disk space?

hirbod, 2022-03-28:
I've double checked, disk space is not the issue, 16GB free space left.

hirbod, 2022-04-14:
@mrousavy  I found a fix for it thanks to this article
https://code-examples.net/en/q/e14f33

PR incoming

closed 2022-04-15 by mrousavy

hirbod, 2022-04-15:
So sorry @mrousavy but I really thought my PR fixed it. It happened again, but less frequent though.

Maybe you'll need to revert my PR

commit 2ccce358 (2022-04-15): fix: AVAssetWriter status is 1 crash (#995)

fixes #930

markAsFinished has to be called within the finishWritingWithCompletionHandler.
Moving that 2 lines allowed me to spam start / stop for hundreds of time without a single crash.