margelo/react-native-vision-camera #931: 🐛 [ios] - Crashes with fileType and videoCodec under specific scenarios

hirbod, 2022-03-24:
### What were you trying to do?

Using the newly introduced `fileType` and `videoCodec` settings (set to mp4 and h264), I receive a lot of crashes from users with iPhone 11 Pro and iPhone SE.

If they try to record a video **WITHOUT** HDR enabled, the app crashes. On iPhone SE 2nd Generation with iOS 15.4, this only happens for the front and back camera (using the device determination provided by the example app)

It looks like that setting for videoCodec is not working with every `device` and setting (FPS, Video Size, HDR, Low Light?).

Using the method to determine available codecs does not help, since it looks like it does not take the current camera settings into consideration afaik.

Removing the codec "fixed" the crash. In my case, providing the codec is not an issue, since the videos will be transcoded with an external SDK anyway which is able to read all the files we're producing, but it might be an issue for folk who really need to have the output as mp4 with h264.

### Reproduceable Code

```tsx
camera.current.startRecording({
          flash: flash,
          onRecordingError: (_error) => {
            //console.error('Recording failed!', error);
            setIsProcessing(false);
            onStoppedRecording();
          },
          onRecordingFinished: (video) => {
            //console.log(`Recording successfully finished! ${video.path}`)
            onMediaCaptured(video, 'video');
            onStoppedRecording();
          },
          fileType: 'mp4',
          videoCodec: 'h264',
        });
```

### What happened instead?

The app crashes with `Invalid video codec type`

### Relevant log output

```shell
NSInvalidArgumentException: *** -[AVCaptureVideoDataOutput recommendedVideoSettingsForVideoCodecType:assetWriterOutputFileType:] Invalid video codec type - available video codec types are: (
    hvc1
)
```

### Device

iPhone 11 Pro, iPhone SE 2nd Generation (15.4)

### VisionCamera Version

2.13.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-03-28:
Does `Camera.getAvailableVideoCodecs()` return h264 for you, or why are you passing that to `videoCodec`?

hirbod, 2022-03-28:
I assumed that h264 is always available. But it is not. Is `getAvailableVideoCodecs` a live query based on device/format etc? If yes, the issue is most likely my own faul, if not, then it is a general issue. IIRC, getAvailableVideoCodecs was not changing for me when I switched between front- and back camera.

I kinda interpreted this more like a "force codec" option.

closed 2022-03-28 by mrousavy

hirbod, 2022-03-28:
In my case, the codec is really not important. But it might be for people who want to produce H264 Videos for web as an example, without being forced to transcode again. Since I am transcoding anyway, I could easily remove the codec.

I think I just misused it. Thanks for clarification. 

Maybe worth a little hint inside the docs. Shall I provide a PR for it later?

mrousavy, 2022-03-28:
No thank you, I'll try to come up with a simpler API soon then we don't need users to think too much 😅