margelo/react-native-vision-camera #934: 🐛 Build Error Android

savataqe1, 2022-03-25:
### What were you trying to do?

when i try to start the project i get metadata error.

### Reproduceable Code

```tsx
ext {
        buildToolsVersion = "30.0.2"
        minSdkVersion = 21
        compileSdkVersion = 30
        targetSdkVersion = 30
        ndkVersion = "21.4.7075529"
    }
```

### What happened instead?

info Running jetifier to migrate libraries to AndroidX. You can disable it using "--no-jetifier" flag.   
Jetifier found 1245 file(s) to forward-jetify. Using 12 workers...
info JS server already running.
info Installing the app...

> Configure project :lottie-react-native  
<==-----------> 22% C> :react-native-sharC> Configure project :react-native-vision-camera
VisionCamera: node_modules/ found at: D:\son\furya-app\node_modules
VisionCamera: Building with Hermes...     
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.      
VisionCamera: Frame Processors are enabled! Building C++ part...
WARNING:: The specified Android SDK Build 
Tools version (28.0.3) is ignored, as it is below the minimum supported version (30.0.2) for Android Gradle Plugin 4.2.2.     
Android SDK Build Tools 30.0.2 will be used.
To suppress this warning, remove "buildToolsVersion '28.0.3'" from your build.gradle file, as each version of the Android Gradle Plugin now has a default version of the build tools.        

The Kotlin Gradle plugin was loaded multiple times in different subprojects, which is not supported and may break the build.  
This might happen in 
subprojects that apply the Kotlin plugins 
with the Gradle 'plugins { ... }' DSL if they specify explicit 
versions, even if the versions are equal. 
Please add the Kotlin plugin to the common parent project or the root project, then 
remove the versions in the subprojects.   
If the parent project does not need the plugin, add 'apply false' to the plugin line.
See: https://docs.gradle.org/current/userguide/plugins.html#sec:subprojects_plugins_dsl
The Kotlin plugin was loaded in the following projects: ':react-native-pager-view', 
':react-native-vision-camera'

> Task :app:checkDebugAarMetadata FAILED  

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.9/userguide/command_line_interface.html#sec:command_line_warnings
26 actionable tasks: 
2 executed, 24 up-to-date
Warning: Mapping new 
ns http://schemas.android.com/repository/android/common/02 to old ns http://schemas.android.com/repository/android/common/01  
Warning: Mapping new 
ns http://schemas.android.com/repository/android/generic/02 to 
old ns http://schemas.android.com/repository/android/generic/01Warning: Mapping new 
ns http://schemas.android.com/sdk/android/repo/addon2/02 to old ns http://schemas.android.com/sdk/android/repo/addon2/01      
Warning: Mapping new 
ns http://schemas.android.com/sdk/android/repo/repository2/02 to old ns http://schemas.android.com/sdk/android/repo/repository2/01
Warning: Mapping new 
ns http://schemas.android.com/sdk/android/repo/sys-img2/02 to old ns http://schemas.android.com/sdk/android/repo/sys-img2/01  
Warning: Mapping new 
ns http://schemas.android.com/repository/android/common/02 to old ns http://schemas.android.com/repository/android/common/01  
Warning: Mapping new 
ns http://schemas.android.com/repository/android/generic/02 to 
old ns http://schemas.android.com/repository/android/generic/01Warning: Mapping new 
ns http://schemas.android.com/sdk/android/repo/addon2/02 to old ns http://schemas.android.com/sdk/android/repo/addon2/01      
Warning: Mapping new 
[... 60 lines cut ...]
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org      
BU�LD FAILED in 16s

    at makeError (D:\son\furya-app\node_modules\@react-native-c\@react-native-community\cli-platform-android\nd\index.js:174:9ode_modules\execa\index.js:174:9)
    at D:\son\furya-app\node_modules\@react-native-community\clive-community\cli-platform-android\node_modulese78:16\execa\index.js:278:16                          process/task_qu
    at processTicksAndRejections (node:internal//process/task_queues:96:5)                      ode_modules\@re
    at async runOnAllDevices (D:\son\furya-app\ncommands\runAndnode_modules\@react-native-community\cli-platfomrm-android\build\commands\runAndroid\runOnAllDecapp\node_modulevices.js:109:5)                                 :9)
    at async Command.handleAction (D:\son\furya-s.-app\node_modules\@react-native-community\cli\biuild\index.js:192:9)

### Device

Pixel 4 XL API 29

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Gabriel8579, 2022-03-25:
Set compileSdkVersion = 31, targetSdkVersion = 31

savataqe1, 2022-03-25:
> Set compileSdkVersion = 31, targetSdkVersion = 31
@Gabriel8579 Thank you very much for your answer but i tried before. This time output.
`Configure project :react-native-vision-camer 
VisionCamera: node_modules/ found at: D:\son\furya-app\node_modules
VisionCamera: Building with Hermes...
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.
VisionCamera: Frame Processors are enabled! Building C++ part...
WARNING:: The specified Android SDK Build Tools version (28.0.3) is ignored, as it is below the minimum supported version (30.0.2) for Android Gradle Plugin 4.2.2.
Android SDK Build Tools 30.0.2 will be used.   
To suppress this warning, remove "buildToolsVersion '28.0.3'" from your build.gradle file, as 
each version of the Android Gradle Plugin now has a default version of the build tools.       

The Kotlin Gradle plugin was loaded multiple times in different subprojects, which is not supported and may break the build.
This might happen in subprojects that apply the Kotlin plugins with the Gradle 'plugins { ... 
}' DSL if they specify explicit versions, even 
if the versions are equal.
Please add the Kotlin plugin to the common parent project or the root project, then remove the versions in the subprojects.
If the parent project does not need the plugin, add 'apply false' to the plugin line.
See: https://docs.gradle.org/current/userguide/plugins.html#sec:subprojects_plugins_dsl       
The Kotlin plugin was loaded in the following projects: ':react-native-pager-view', ':react-native-vision-camera'

> Task :app:processDebugMainManifest FAILED    

See http://g.co/androidstudio/manifest-merger for more information about the manifest merger. 

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.   
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.9/userguide/command_line_interface.html#sec:command_line_warnings
154 actionable tasks: 154 executed
Warning: Mapping new ns http://schemas.android.com/repository/android/common/02 to old ns http://schemas.android.com/repository/android/common/01
Warning: Mapping new ns http://schemas.android.com/repository/android/generic/02 to old ns http://schemas.android.com/repository/android/generic/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/addon2/02 to old ns http://schemas.android.com/sdk/android/repo/addon2/01Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/repository2/02 to old ns http://schemas.android.com/sdk/android/repo/repository2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/sys-img2/02 to old ns http://schemas.android.com/sdk/android/repo/sys-img2/01
D:\son\furya-app\android\app\src\main\AndroidManifest.xml Error:
        Apps targeting Android 12 and higher are required to specify an explicit value for `android:exported` when the corresponding component has an intent filter defined. See https://developer.android.com/guide/topics/manifest/activity-element#exported for details.

FAILURE: Build failed with an exception.       

* What went wrong:
Execution failed for task ':app:processDebugMainManifest'.
> Manifest merger failed : Apps targeting Android 12 and higher are required to specify an explicit value for `android:exported` when the corresponding component has an intent filter defined. See https://developer.android.com/guide/topics/manifest/activity-element#exported for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get 
more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org     

BU�LD FAILED in 17s

error Failed to install the app. Make sure you 
have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081
Warning: Mapping new ns http://schemas.android.com/repository/android/common/02 to old ns http://schemas.android.com/repository/android/common/01
Warning: Mapping new ns http://schemas.android.com/repository/android/generic/02 to old ns http://schemas.android.com/repository/android/generic/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/addon2/02 to old ns http://schemas.android.com/sdk/android/repo/addon2/01Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/repository2/02 to old ns http://schemas.android.com/sdk/android/repo/repository2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/sys-img2/02 to old ns http://schemas.android.com/sdk/android/repo/sys-img2/01
D:\son\furya-app\android\app\src\main\AndroidManifest.xml Error:
        Apps targeting Android 12 and higher are required to specify an explicit value for `android:exported` when the corresponding component has an intent filter defined. See https://developer.android.com/guide/topics/manifest/activity-element#exported for details.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:processDebugMainManifest'.
[... cut ...]
    at D:\son\furya-app\node_modules\@react-native-community\cli-platform-android\node_modules\execa\index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (D:\son\furya-app\node_modules\@react-native-community\cli-platform-android\build\commands\runAndroid\runOnAllDevices.js:109:5)
    at async Command.handleAction (D:\son\furya-app\node_modules\@react-native-community\cli\build\index.js:192:9)
info Run CLI with --verbose flag for more details.`

Gabriel8579, 2022-03-25:
Just set android:exported "true" in your MainActivity (Android Manifest) as shown [https://stackoverflow.com/questions/67412084/android-studio-error-manifest-merger-failed-apps-targeting-android-12](here)

savataqe1, 2022-03-25:
my settings are already like this ;
```
<activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize|uiMode"
        android:launchMode="singleTask"
        android:windowSoftInputMode="adjustResize"
        android:exported="true" />
```

Gabriel8579, 2022-03-25:
```
D:\son\furya-app\android\app\src\main\AndroidManifest.xml Error:
        Apps targeting Android 12 and higher are required to specify an explicit value for `android:exported` when the corresponding component has an intent filter defined. See https://developer.android.com/guide/topics/manifest/activity-element#exported for details.
```

kabitacode, 2022-03-28:
> ```
> D:\son\furya-app\android\app\src\main\AndroidManifest.xml Error:
>         Apps targeting Android 12 and higher are required to specify an explicit value for `android:exported` when the [...]

same error, did you finish it?

savataqe1, 2022-03-28:
@kabitacode not yet, still trying to solve it .

mrousavy, 2022-03-28:
Create a reproduceable example app. If you add `android:exported="true"`, you shouldn't be getting the error you posted. Maybe clean and rebuild.

Also, there are multiple similar issues, so I don't believe you that you searched the repo.

[image]

closed 2022-03-28 by mrousavy

iKronyck, 2022-06-09:
Same error :/

ameer-taghavi, 2022-06-28:
set:
```
buildToolsVersion = "30.0.2"
compileSdkVersion = 31
targetSdkVersion = 30
```

jasbirrajrana, 2022-10-03:
@mrousavy why do we need to put android:exported="true" after adding this I received security concerns from Google VAPT.
any alternative solution to this?

ArturoTorresMartinez, 2022-10-29:
Same issue, all other issues on the repo have suggestions on how to solve but they have not worked for me either