margelo/react-native-vision-camera #936: 🐛 [ios] Crashing on fast start / stopRecording calls - Unexpectedly found nil while unwrapping an Optional value

hirbod, 2022-03-25:
### What were you trying to do?

When I try to create couple of short vide shots (<1s) with a more rapid "tapping" approach, older devices hick up and crash.
It does not happen on powerful iOS devices, but iPhone 7 and SE tend to crash quite frequently, even if I wait a bit more before i hit the button

### Reproduceable Code

```tsx
Just called startRecord / stopRecording quite frequently. Will happen at some point (sometimes after the 5th try, sometimes after the 15th hit).
```

### What happened instead?

The app crashes with `Thread 40: Fatal error: Unexpectedly found nil while unwrapping an Optional value`

[image]
`

### Relevant log output

```shell
The app crashes with `Thread 40: Fatal error: Unexpectedly found nil while unwrapping an Optional value
```

### Device

iPhone 7 (iOS 12.3.1)

### VisionCamera Version

2.13.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

hirbod, 2022-03-25:
I think assuming its not `nil` is prolly not a good idea.

https://github.com/mrousavy/react-native-vision-camera/blob/main/ios/CameraView%2BRecordVideo.swift#L125
https://github.com/mrousavy/react-native-vision-camera/blob/main/ios/CameraView%2BRecordVideo.swift#L135

The question is: should we wrap it with do/catch and reject the callback or is there any other way to make sure this does not happen?

I know that calling activateAudioSession/deactivateAudioSession frequently is a bad thing. I've experienced similar issues with expo-av and we decided not to deactivate any audiosessions instead (since it is a blocking task and prone to errors).

We decided to disable it here and my PR got merged. Maybe this is related. Because I am hitting start/stop so often, it starts the audiosession and stops the audiosession.

Maybe the audiosession should only be killed when camera is unmounted/not active?
https://github.com/expo/expo/pull/16578#issuecomment-1072732278

usaidather, 2022-03-28:
@mrousavy when will you push this issue to live, as this happening with me as well.

closed 2022-04-15 by mrousavy

vntech-engineering69, 2023-03-17:
Hello, 

Is this issue fixed ?
I am using version 2.15.4.
The camera stops on iPhone 12.
Is there any temporary fix ?

[image]

vntech-engineering69, 2023-03-19:
@hirbod Can you please check into this. Thank you.

hirbod, 2023-03-19:
@vntech-engineering69 even though the errors sound similar, they are unrelated. I'd recommend opening a new issue for this. Not sure if you get much attention though, since v3 is around the corner as far as I can tell

PR #938 (2022-04-15): fix: Fix RecordingSession nil crash by keeping it local
Fixes #936