margelo/react-native-vision-camera #937: ✨ OCR support implementd?

jolic, 2022-03-27:
### What feature or enhancement are you suggesting?

I looked at https://github.com/mrousavy/react-native-vision-camera/issues/313 but I cannot figure out how to use that )if this is implemented).

If it's implemented, could someone please paste a code  or some infos how to use that?

### What Platforms whould this feature/enhancement affect?

iOS

### Alternatives/Workarounds

Use other solutions like:
https://github.com/agoldis/react-native-mlkit-ocr

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2022-03-28:
Hey, please read the Frame Processor documentation. What you're looking for should not be a core feature of a camera library, but rather a Frame Processor Plugin.

mrousavy, 2022-03-28:
Also, that's the same thing I said in #313.

closed 2022-03-28 by mrousavy