margelo/react-native-vision-camera #945: 🐛  google pixel 4a, has night sight, though lib returns supportsLowLightBoost: false

nikoloz110, 2022-03-28:
### What were you trying to do?

I want to achieve the same effect with vision camera, as when I turn on night sight on my native google pixel 4a camera. but the vision camera api tells me that the wide-camera (back), supportsLowLightBoost: false. is this some kind of bug, or just a different thing?

### What happened instead?

supportsLowLightBoost: false, I think it should return true, since my camera has, night sights

### Device

google pixel 4a

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

mrousavy, 2023-08-18:
Maybe the low light boost feature is an extension exposed through Camera2 only, CameraX only, or not exposed at all.

closed 2023-09-01 by mrousavy