margelo/react-native-vision-camera #952: 🐛 Frame processor crashed application

vgorte, 2022-03-30:
### What were you trying to do?

I wanted to use the vision-camera with the [code-scanner](https://github.com/rodgomesc/vision-camera-code-scanner) plugin. 
All installations steps (vision-camera, code-scanner, reanimated) have been executed by me, like instructed on the corresponding installation guides.

### Reproduceable Code

```tsx
Component: 
export const QRScannerScreen = () => {
  const [ hasPermission, setHasPermission ] = React.useState(false);
  const devices = useCameraDevices();
  const device = devices.back;

  const [ frameProcessor, barcodes ] = useScanBarcodes([ BarcodeFormat.QR_CODE ]);

  React.useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'authorized');
    })();
  }, []);

  if (!device || !hasPermission) return <Text>Fallback</Text>;

  return (hasPermission && device &&  (
    <>
      <Camera
        style={ StyleSheet.absoluteFill }
        device={ device }
        isActive={ true }
        frameProcessor={ frameProcessor }
        frameProcessorFps={ 1 }
      />
      { barcodes.map((barcode, idx) => (
        <Text key={ idx } style={ styles.barcodeTextURL }>
          { barcode.displayValue }
        </Text>
      )) }
    </>
  ));
};

babel.config.js:
module.exports = {
  presets: ['module:metro-react-native-babel-preset'],
  plugins: [
    [
      'react-native-reanimated/plugin',
      {
        globals: ['__scanCodes'],
      },
    ]
  ],
};
```

### What happened instead?

The `camera` component is working like intended. However, as soon as I pass the `frameProcessor` property, the application crashes with an `java.lang.UnsatisfiedLinkError`

I searched for other issues with this kind of error but could not find any / was not able to fix the error with the provided solutions.

### Relevant log output

```shell
FATAL EXCEPTION: pool-10-thread-1

java.lang.UnsatisfiedLinkError: No implementation found for void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy) (tried Java_com_mrousavy_camera_CameraView_frameProcessorCallback and Java_com_mrousavy_camera_CameraView_frameProcessorCallback__Landroidx_camera_core_ImageProxy_2)

FATAL EXCEPTION: pool-10-thread-1
Process: XX.XX.XX.XXXXXXXX, PID: 13593
java.lang.UnsatisfiedLinkError: No implementation found for void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy) (tried Java_com_mrousavy_camera_CameraView_frameProcessorCallback and Java_com_mrousavy_camera_CameraView_frameProcessorCallback__Landroidx_camera_core_ImageProxy_2)
	at com.mrousavy.camera.CameraView.frameProcessorCallback(Native Method)
	at com.mrousavy.camera.CameraView.configureSession$lambda-7$lambda-6(CameraView.kt:484)
	at com.mrousavy.camera.CameraView.$r8$lambda$cqtIchEZdTZaV3R0UUrDpVbB1Es(Unknown Source:0)
	at com.mrousavy.camera.CameraView$$ExternalSyntheticLambda1.analyze(Unknown Source:2)
	at androidx.camera.core.ImageAnalysis.lambda$setAnalyzer$2(ImageAnalysis.java:463)
	at androidx.camera.core.ImageAnalysis$$ExternalSyntheticLambda0.analyze(Unknown Source:2)
	at androidx.camera.core.ImageAnalysisAbstractAnalyzer.lambda$analyzeImage$0$androidx-camera-core-ImageAnalysisAbstractAnalyzer(ImageAnalysisAbstractAnalyzer.java:283)
	at androidx.camera.core.ImageAnalysisAbstractAnalyzer$$ExternalSyntheticLambda1.run(Unknown Source:14)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:923)
```

### Device

OnePlus 6t (Android 11)

### VisionCamera Version

2.13.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-03-31:
Maybe this is a Proguard issue? Can you try disabling Proguard?

vgorte, 2022-03-31:
I have proguard disabled. The issue still persists.
I would happily provide more logs, if you could tell me what kind of logs would be helpful.

vgorte, 2022-03-31:
I shared the code with someone else and he was able to build and run the aplpication + frame processor.
That would indicate a problem with my setup, I guess. 
We checked the node and java versions, which matched between his computer and mine. 

I think the installations of reanimated and vision-camera are correct, since he was able to simply install and run it.

Is there anything else in my environment that could cause this issue?

vgorte, 2022-04-08:
I managed to resolve the issue. It looked like none of the c code was compiling. It looked like CMake was not correctly installed or something of that sort. Works fine now

closed 2022-04-08 by vgorte

kaiyes, 2022-07-20:
> I managed to resolve the issue. It looked like none of the c code was compiling. It looked like CMake was not correctly installed or something of that sort. Works fine now

This seems to solve it for this dev