margelo/react-native-vision-camera #954: App crashes in release mode🐛

palagonKevo, 2022-03-31:
### What were you trying to do?

I'm trying to assemble the release apk.

### Reproduceable Code

```tsx
Just cd android/ and then ./gradlew assembleRelease.

This are the dependencies:

node: v16.13.0
react-native: 6.4.0
os: Ubuntu 21.10
react-native-vision-camera: ^2.13.1
vision-camera-dynamsoft-barcode-reader: ^0.3.1

babel config:
module.exports = {
  presets: ['module:metro-react-native-babel-preset'],
  plugins: [
    ["module:react-native-dotenv", {
      "moduleName": "@env",
      "path": ".env",
      "blacklist": null,
      "whitelist": null,
      "safe": false,
      "allowUndefined": true
    }],
    [
      'react-native-reanimated/plugin',
      {
        globals: ['__decode'],
      },
    ]
  ],
  env: {
    production: {
      plugins: ['react-native-paper/babel','react-native-reanimated/plugin'],
    },
  },
};
```

### What happened instead?

 When it's built, and when trying to use the camera, the apk crashes every time. This works perfectly on debug mode.

### Relevant log output

```shell
This is what I get from adb logcat *:S ReactNative:V ReactNativeJS:V.

03-31 09:50:25.553 18158 18187 E ReactNativeJS: ReferenceError: Property '__decode' doesn't exist, js engine: hermes
03-31 09:50:25.554 18158 18187 E ReactNativeJS: TypeError: Cannot read property 'decode' of undefined
03-31 09:50:25.554 18158 18187 E ReactNativeJS: 
03-31 09:50:25.554 18158 18187 E ReactNativeJS: This error is located at:
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in v
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RNSScreen
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in R
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in D
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in B
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in h
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RNSScreenStack
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in ScreenStack
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in y
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RNCSafeAreaProvider
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in y
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in D
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in l
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in b, js engine: hermes
03-31 09:50:25.555 18158 18187 E ReactNativeJS: TypeError: Cannot read property 'decode' of undefined
03-31 09:50:25.555 18158 18187 E ReactNativeJS: 
03-31 09:50:25.555 18158 18187 E ReactNativeJS: This error is located at:
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in v
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RNSScreen
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in R
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in D
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in B
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in h
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RNSScreenStack
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in ScreenStack
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in y
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RNCSafeAreaProvider
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in y
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in D
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in l
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in b, js engine: hermes
```

### Device

Mi Note 10 Pro

### VisionCamera Version

2.13.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-03-31:
Hey, can you share your function `decode` and your `useFrameProcessor`?

palagonKevo, 2022-03-31:
I've followed this steps: https://www.dynamsoft.com/codepool/react-native-qr-code-scanner-vision-camera.html
There's all I've done this far.

mrousavy, 2022-03-31:
did you add the native plugin as mentioned in that article? it looks to me like the native plugin is not correctly registered.
As I said earlier, please share your code. I cannot see your screen from where I'm sitting.

Also, please share the initial logs that are logged by VisionCamera to Logcat on app-start, it says "Registering Plugins" or something like that.

 maybe @xulihang has an idea?

xulihang, 2022-03-31:
This seems to be related to proguard. Have you enabled proguard?

Try to add the following rules:

```
-keep class com.swmansion.reanimated.** { *; }
-keep class com.facebook.react.turbomodule.** { *; }
-keep class com.dynamsoft.dbr.** { *; }
```

xulihang, 2022-03-31:
Try to disable proguard first. If the error still exists, it is not related to proguard.

xulihang, 2022-03-31:
In my case, when I run the [example](https://github.com/xulihang/vision-camera-dynamsoft-barcode-reader/example) with proguard disabled, it works fine.

But when I enabled proguard, there is a worklet error even after I have added proguard  rules.

palagonKevo, 2022-03-31:
Seem to be related with proguard, but I need it if I wanna shared this app.

xulihang, 2022-03-31:
Vision camera may also need adding rules. I find a commit about this: https://github.com/mrousavy/react-native-vision-camera/pull/465/files

palagonKevo, 2022-03-31:
I've just found that one as well, it's building, let see if that works. .

palagonKevo, 2022-03-31:
Na, I'm trying this rules:
```
-keep class com.swmansion.reanimated.** { *; }
-keep class androidx.camera.extensions.** {*;}
-keep class com.facebook.react.turbomodule.** { *; }
-keep class com.dynamsoft.dbr.** { *; }
-keep class com.facebook.hermes.unicode.** { *; }
-keep class com.facebook.jni.** { *; }
```
But nothing seems to be solving the problem.

xulihang, 2022-04-01:
I've managed to solve this by adding the rules mentioned in this Xamarin project: https://github.com/JimmyPun610/BarcodeScanner.XF

The following rule is also needed:

```
-keep class androidx.camera.core.** {*;}
```

palagonKevo, 2022-04-01:
Same error with crash.

xulihang, 2022-04-01:
What is the error message? My build works fine with the following rules:

```
-keep class com.swmansion.reanimated.** { *; }
-keep class com.facebook.react.turbomodule.** { *; }
-keep class com.dynamsoft.dbr.** { *; }
-keep class androidx.camera.core.** {*;}
```

palagonKevo, 2022-04-01:
The error is the same as the last one, like this one:
```
03-31 09:50:25.553 18158 18187 E ReactNativeJS: ReferenceError: Property '__decode' doesn't exist, js engine: hermes
03-31 09:50:25.554 18158 18187 E ReactNativeJS: TypeError: Cannot read property 'decode' of undefined
03-31 09:50:25.554 18158 18187 E ReactNativeJS: 
03-31 09:50:25.554 18158 18187 E ReactNativeJS: This error is located at:
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in v
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RNSScreen
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in R
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in D
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in B
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in h
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RNSScreenStack
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in ScreenStack
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in y
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RNCSafeAreaProvider
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in y
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in D
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in l
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.554 18158 18187 E ReactNativeJS:     in b, js engine: hermes
03-31 09:50:25.555 18158 18187 E ReactNativeJS: TypeError: Cannot read property 'decode' of undefined
03-31 09:50:25.555 18158 18187 E ReactNativeJS: 
03-31 09:50:25.555 18158 18187 E ReactNativeJS: This error is located at:
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in v
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RNSScreen
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in R
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in D
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in B
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in h
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RNSScreenStack
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in ScreenStack
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in y
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RNCSafeAreaProvider
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in y
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in D
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in l
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in RCTView
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in Unknown
03-31 09:50:25.555 18158 18187 E ReactNativeJS:     in b, js engine: hermes
```

xulihang, 2022-04-01:
Then I have no idea what the reason is. Maybe you can examine your project with my example: https://github.com/xulihang/vision-camera-dynamsoft-barcode-reader/tree/main/example

palagonKevo, 2022-04-01:
It seem that when minifying the project, the apk doesn't recognize the decode function, the one included here:
`import { DBRConfig, decode, TextResult } from 'vision-camera-dynamsoft-barcode-reader'
`
An after I try to open the camera, it crashes. That's what I understood from the error, which is kinda strange because it works as it should with the debug mode.

mrousavy, 2022-04-04:
Minifying? Are you using a custom minifyer? That might be the reason because it renames functions like `__decode(..)` to `a(..)`!

palagonKevo, 2022-04-04:
By minify I mean this 
```
release {
   ...
    minifyEnabled enableProguardInReleaseBuilds
    proguardFiles getDefaultProguardFile("proguard-android.txt"), "proguard-rules.pro"
}
```

iskezil, 2022-12-20:
Is there a solution to the problem?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

ChetanAlmelkar, 2023-10-23:
If I used enableProguardInReleaseBuilds=true then app crashes when i open camera from react-native-vision-camera.

mrousavy, 2023-10-23:
Does it work if you disable proguard?