margelo/react-native-vision-camera #959: "[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames. One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated."🐛

clovisrodriguez, 2022-04-01:
### What were you trying to do?

I'm trying to create blob from the file output from the camera, by fetching the path produced by the `VideoFile`interface, once I successfully created the blob and uploaded it to our VOD provider, then when I try to shoot a new file I'm getting this error after I navigate to the video replay component.

### Reproduceable Code

```tsx
import React, { useEffect, useRef, useState } from 'react';
import { PERMISSIONS, check } from 'react-native-permissions';
import { Linking, StyleSheet } from 'react-native';
import { useSelector } from 'react-redux';
import { Button, Text, View, TouchableOpacity } from 'react-native-ui-lib';
import { Camera, useCameraDevices } from 'react-native-vision-camera';
import { useIsFocused } from '@react-navigation/native';
import Icon from 'react-native-vector-icons/MaterialCommunityIcons';

import { RootState } from '@store';
import { Theme, themeColors } from '@style';
import { useIsForeground } from '@hooks';
import { ROUTES } from '../../components/Router/Router';
import { style } from './CameraScreen.style';
import { Timer } from './Timer';

enum CAMERA_PERMISSIONS {
  AUTHORIZED = 'authorized',
  NOT_DETERMINED = 'not-determined',
  DENIED = 'denied',
  RESTRICTED = 'restricted',
}

export const CameraScreen = ({ navigation }) => {
  const [cameraIsReady, setCameraIsReady] = useState(false);
  const [isRecording, setIsRecording] = useState(false);
  const [flash, setFlash] = useState<any>('off');
  const devices = useCameraDevices();
  const [lens, setLens] = useState('back');
  const device = devices[lens];
  const camera = useRef<Camera>(null);
  const isActive = useIsForeground() && useIsFocused();
  const { language } = useSelector((state: RootState) => state);

  const checkPermissions = async () => {
    const cameraPermission = await Camera.getCameraPermissionStatus();
    const microphonePermission = await Camera.getMicrophonePermissionStatus();
    const storagePermissions = await check(PERMISSIONS.ANDROID.READ_EXTERNAL_STORAGE);

    if (
      cameraPermission === CAMERA_PERMISSIONS.AUTHORIZED &&
      microphonePermission === CAMERA_PERMISSIONS.AUTHORIZED
    ) {
      setCameraIsReady(true);
    } else if (
      cameraPermission === CAMERA_PERMISSIONS.NOT_DETERMINED &&
      microphonePermission === CAMERA_PERMISSIONS.NOT_DETERMINED
    ) {
      const newCameraPermission = await Camera.requestCameraPermission();
      const newMicrophonePermission = await Camera.requestMicrophonePermission();

      if (
        newCameraPermission === CAMERA_PERMISSIONS.AUTHORIZED &&
        newMicrophonePermission === CAMERA_PERMISSIONS.AUTHORIZED
      )
        setCameraIsReady(true);
    }
  };

  const onPressRecording = async () => {
    const stopRecording = async () => {
      if (camera.current) await camera.current.stopRecording();
      setIsRecording(false);
      return;
    };

    if (isRecording) {
      stopRecording();
    }

    setIsRecording(true);

    if (camera.current)
      camera.current.startRecording({
        onRecordingFinished: (video) => {
          const blob = await fetch(video.path)
          navigation.navigate(ROUTES.VIDEO_PLAYER, { video, path })
        },
        onRecordingError: (error) => console.error(error),
      });
  };

  const closeCamera = () => {
    navigation.navigate(ROUTES.HOME);
  };

  useEffect(() => {
    checkPermissions();
  }, []);

  const checkCamera = () => {
    return device ? (
      <React.Fragment>
        <Camera
          ref={camera}
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isActive}
          preset="medium"
          video={true}
          torch={flash}
        />
        <View style={style.cameraActionsContainer}>
          <TouchableOpacity style={style.buttonContainer} onPress={closeCamera}>
            <Text>
              <Icon name="close" size={24} color="#fff" />
            </Text>
          </TouchableOpacity>
          <TouchableOpacity
            style={style.buttonContainer}
            onPress={() => (lens === 'front' ? setLens('back') : setLens('front'))}
          >
            <Text>
              <Icon name="camera-switch" size={24} color="#fff" />
            </Text>
          </TouchableOpacity>
          <TouchableOpacity
            style={style.buttonContainer}
            onPress={() => (flash === 'off' ? setFlash('on') : setFlash('off'))}
          >
            <Text>
              <Icon name={flash === 'off' ? 'flash-off' : 'flash'} size={24} color="#fff" />
            </Text>
          </TouchableOpacity>
        </View>
        <View style={style.cameraActionsBottomContainer}>
          {isRecording && <Timer />}
          <TouchableOpacity style={style.buttonContainerRecord} onPress={onPressRecording}>
            <Text>
              <Icon
                name={isRecording ? 'square-rounded' : 'record'}
                size={50}
                color={themeColors.errorColor}
              />
            </Text>
          </TouchableOpacity>
        </View>
      </React.Fragment>
    ) : null;
  };

  return cameraIsReady ? (
    checkCamera()
  ) : (
    <View style={style.container}>
      <View center padding-s8>
        <Text s3 primaryColor center>
          {language.CAMERA_SCREEN.TITLE_PERMISSIONS}
        </Text>
        <Text body center marginV-s8>
          {language.CAMERA_SCREEN.BODY_PERMISSIONS}
        </Text>
        <Button
          style={Theme.mdButton}
          labelStyle={Theme.labelStyleButton}
          label={language.CAMERA_SCREEN.LABEL_PERMISSIONS}
          onPress={() => Linking.openSettings()}
        />
      </View>
    </View>
  );
};
```

### What happened instead?

```
{
  "message": "[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames. One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.",
  "stacktrace":  "com.mrousavy.camera.InactiveSourceError: [capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.
	at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:64)
	at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:52)
	at androidx.camera.video.Recorder$RecordingRecord.lambda$updateVideoRecordEvent$6$Recorder$RecordingRecord(Recorder.java:2624)
	at androidx.camera.video.-$$Lambda$Recorder$RecordingRecord$Uo40MCKPuma4aZ5df1-_fzufzHg.run(Unknown Source:4)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:236)
	at android.app.ActivityThread.main(ActivityThread.java:8025)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:620)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1011)
"
}
```

### Device

Xiaomi redmi note 8

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

gev3, 2022-06-01:
Same issue :(

trung24012001, 2022-06-14:
Is there anyone resolve this issue ?

juanpadin, 2022-07-13:
same issue

clovisrodriguez, 2022-07-13:
Well sorry, I didn't give an update earlier, but this is probably happening because you are in debugging mode, after I updated the project for using the Hermes engine and stopped using the debugging mode it never happened again.

I hope it works for everybody that has this issue.

Aakash359, 2022-10-12:
I am facing the same... The recording failed because the source becomes inactive and stops sending frames. One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated, Please do let me know any solutions of this problem

dziter, 2022-12-05:
I am facing with the same issue when I flip the camera during recording on Android.
On iOS, everything is fine.

ratz6, 2023-02-21:
any fix for this ?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy