margelo/react-native-vision-camera #960: 🐛 Frame is sideways on android portrait orientation

jslok, 2022-04-01:
### What were you trying to do?

When using frame processor on android in portrait mode, the frame is rotated to the left. When rotated to landscape mode, the frame is the correct orientation. 

Orientation prop from #715 does not seem to have an effect here. This prop is not mentioned in the docs either.

### What happened instead?

Frame should be the correct orientation in portrait.

### Device

galaxy s20

### VisionCamera Version

2.12.0 and 2.13.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jslok, 2022-04-01:
Here's screenshots of what I mean. In each screenshot, the top is the camera live view and the bottom is the frame from frameprocessor.
![277245470_569968350731650_9191431795062350398_n](https://user-images.githubusercontent.com/22921326/161240098-f0608cdf-44aa-44f8-a862-b726e5a
[image]
eaec7.jpg)

mrousavy, 2022-04-04:
Hey, thanks for the detailed issue and explanation with screenshots.

You're right, this is actually a big issue, the Frame should be oriented correctly - does the `Frame.orientation` value change at all or is it `portrait` in every case?

mrousavy, 2022-04-04:
I think this is the native orientation of the frame, since Camera sensors are in landscape orientation by default. If I would actually rotate the frame (moving each pixel to it's rotated value), this would be really really slow because I have to do that every frame.

If the `Frame.orientation` value is correct, there's not much more to do since this is what is expected by all frame processing libraries like MLKit Vision (they all have a `orientation` flag that tells them how to interpret the frame)

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy