margelo/react-native-vision-camera #965: ❓ Support for Live Push/Publish and Play Functions of Live Streaming like react-native-nodemediaclient

alvintheodora, 2022-04-05:
### Question

Hi developers, 
I am wondering if this library support for Live Streaming feature, in which can act to Push Live Stream and also Play Live Stream via RTMP, like https://github.com/NodeMedia/react-native-nodemediaclient of `NodeCameraView` and `NodePlayerView`.
If yes, how should it be done?

Thank you in advance.
Cheers.

### VisionCamera Version

2.13.1 or latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-04-06:
Hey, yes you can send video frames to your webserver (WebRTC) using a custom Frame Processor Plugin. Read more about [creating Frame Processor Plugins here](https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processors-plugins-overview).

And to your second question; no, Vision**Camera** does not have a video player and will never have one.

closed 2022-04-06 by mrousavy

alvintheodora, 2022-04-11:
Hi @mrousavy , thanks for clarifying. Are there any ready-plugins for the case of "send video frames to your webserver (WebRTC)"?
And how about RTMP, does it also support?

mrousavy, 2022-04-11:
@alvintheodora 

>  there any ready-plugins for the case of "send video frames to your webserver (WebRTC)"?

I haven't built any. If you want to pay for the development, I can build one

endrits079, 2022-05-31:
> Hi @mrousavy , thanks for clarifying. Are there any ready-plugins for the case of "send video frames to your webserver (WebRTC)"? And how about RTMP, does it also support?

What solution did you came up with?

shawarmaz, 2022-11-18:
@mrousavy did there end up being any progress on this? 

If not, how much $ to complete? Maybe crowdfund?

mrousavy, 2022-11-21:
Hey! I did not work on this, no. We can crowdfund this, I'm thinking a stable plugin would cost around $5.000

benjipott, 2024-03-08:
I will try to create a plugin that can publish to mux.

giantslogik, 2024-03-21:
@benjipott Do let us know if you build a plugin to publish to mux.  It would be useful in an App i'm working on.

adamgarcia4, 2024-03-22:
@benjipott Seconded!

bilaltahseen, 2024-12-13:
Hi, does anyone come up with a plugin ?