margelo/react-native-vision-camera #966: ❓ How to prevent the flip/mirror?

pierroo, 2022-04-05:
### Question

It feels weird when in selfie mode the resulting image is mirrored / flipped.
When you check the main app (snapchat / tiktok), they don't do the flip, to not distract / confuse the user.
Is there any way to cancel the flip?

### VisionCamera Version

^2.13.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

pierroo, 2022-04-06:
Actually no; just using the vision camera example, we can clearly see that from the time we press the button capture, and move to the mediapage, the mediapage displays a "mirrored" image of the one I saw in real time from the camera page.

This flip is also mentioned here (#130) , although my issue is not the same.
I just want to get rid of this flipping function, users feel naturally "less attractive" when they are being displayed in mirrored image since they are not used to see this when opening their camera.

(just try taking a photo with your hands raised at the left of your face, from camerapage to media page you will see your hands at the other side of the photo)

mrousavy, 2022-04-06:
can you please share a screen recording? Are you talking about Android or iOS?

pierroo, 2022-04-06:
It's happening on android, haven't tested iOS yet.
Will share a screenshot in the next few hours if needed

pierroo, 2022-04-06:
There it is, first is camerapage screenshot, second is mediapage; we can clearly see everything is mirrored from mediapage, thus from the photo generated.
Same happens with video unfortunately
[image]
[image]
.

siddharth-kt, 2022-04-10:
Hi @pierroo , did you got any solution ?

pierroo, 2022-04-10:
Unfortunately none, @mrousavy if you happen to pass by did you have the chance to look into it ?

mrousavy, 2022-04-11:
No, I haven't. Does this happen on iOS as well?

pierroo, 2022-04-11:
> No, I haven't. Does this happen on iOS as well?

Surprisingly no, it does not.
Only happens on android.

mrousavy, 2022-04-11:
Okay yea I think I recall it being an Android issue / CameraX limitation...

pierroo, 2022-04-11:
Thank you for your additional input.
Does that mean it's a lost cause for now, or is there some ongoing investigation about this issue?
So we can know where to stand and whether or not we should adjust our release timeline :)
Thank you!

mrousavy, 2022-04-11:
I currently don't have enough free time to fix this issue, but I am also pretty sure that CameraX does not support this out of the box so this comes with a performance penalty.

amaljosea, 2022-04-14:
@mrousavy I can give it a try to solve this if you could provide some hints.

anatoleblanc, 2022-05-30:
Any updates on this ?

DonDes17, 2022-08-30:
Same problem here with version **2.14.0**
Any idea ?

jpike88, 2022-09-07:
@mrousavy if you look at the iOS camera app, front camera video displays it mirrored on screen, but output video is horizontally flipped. Otherwise, things like words will be back to front.

jpike88, 2022-09-07:
This problem occurs on iOS too (front camera only).

jpike88, 2022-09-28:
I've implemented an easy solution for ios:

CameraView+RecordVideo

```
// get pixel format (420f, 420v, x420)
      let pixelFormat = CMFormatDescriptionGetMediaSubType(videoInput.device.activeFormat.formatDescription)
      recordingSession.initializeVideoWriter(withSettings: videoSettings,
                                                                    pixelFormat: pixelFormat, isFrontCamera: videoInput.device.position == .front)
```

And in RecordingSession:

```
 func initializeVideoWriter(withSettings settings: [String: Any], pixelFormat: OSType, isFrontCamera: Bool) {
    guard !settings.isEmpty else {
      ReactLogger.log(level: .error, message: "Tried to initialize Video Writer with empty settings!")
        return;
    }
    guard bufferAdaptor == nil else {
      ReactLogger.log(level: .error, message: "Tried to add Video Writer twice!")
        return;
    }

    let videoWriter = AVAssetWriterInput(mediaType: .video, outputSettings: settings)
    videoWriter.expectsMediaDataInRealTime = true
        
    if (isFrontCamera) {
      var transform = CGAffineTransform(scaleX: -1.0, y: 1.0);
      videoWriter.transform = transform
    }
```

alexhernandez, 2023-02-09:
@pierroo @amaljosea I noticed that if you initialize the camera with `back` camera instead of `front` camera on init that helps resolve the horizontal flip seen on Android.

ZakirBangash, 2023-05-11:
@mrousavy is this issue resolved in new version ?

FouadMagdy01, 2023-09-26:
Did anyone found a solution for this issue?

JamesHemery, 2024-02-13:
Any solution ?

mrousavy, 2024-07-02:
Hey all! I just spent some time to build a new feature: I created a PR to add a `isMirrored` prop to Camera: https://github.com/mrousavy/react-native-vision-camera/pull/3036

With this prop you can enable or disable output mirroring 🎉🥳

If you appreciate my time and dedication towards building new features and constantly improving this library, please [**consider sponsoring me on GitHub**](https://github.com/sponsors/mrousavy/)  :)

closed 2024-07-02 by mrousavy