margelo/react-native-vision-camera #970: Can't find variable: __scanQRCodes ❓

yamila-fraiman, 2022-04-06:
### Question

I'm getting "Can't find variable: __scanQRCodes"

babel.config.js

```
module.exports = {
	presets: ['module:metro-react-native-babel-preset'],
	plugins: [
		'react-native-paper/babel',
		[
			'react-native-reanimated/plugin',
			{
				globals: ['__scanQRCodes'],
			},
		],
	],
};

```

### What I tried

import {scanQRCodes} from 'vision-camera-qrcode-scanner';

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-04-07:
I cannot help you if I don't have enough information.

mrousavy, 2022-04-07:
Also, you are reporting a **bug**, not asking a question. You used the wrong issue template.

mrousavy, 2022-04-07:
Create a new issue and make sure to fill out the template (I need adb logs, I need REA and RN versions, I need reproduceable code)

closed 2022-04-07 by mrousavy

hengkx, 2022-07-26:
0.69.3 same error

omerts, 2022-08-05:
I had the same issue and fixed with yarn start --reset-cache, or npm start reset-cache

slolkunchik, 2022-08-16:
> yarn start --reset-cache, or npm start reset-cache
doesn't help

hokimtam, 2022-08-29:
/open

tsipiniuk, 2022-09-14:
I solved the problem by downgrading XCode and Command Line Tool to 13.4.1 from 14.0
"react": "18.0.0",
"react-native": "0.69.3",
"react-native-vision-camera": "^2.14.1",
"vision-camera-code-scanner": "^0.2.0"

LazyAfternoons, 2022-09-20:
I'm having the same issue but with ```__decode```. Is downgrading XCode the only way to make it work?

edritech93, 2022-09-26:
same issue for me, I use Xcode 14

victorgalvez56, 2022-09-27:
same issue for me, I use Xcode 14

nurhamsah1998, 2022-12-28:
> I had the same issue and fixed with yarn start --reset-cache, or npm start reset-cache

i try this, and doesn't work

tsipiniuk, 2023-01-25:
upgrade react-native-vision-camera to  2.15.2 version helped me