margelo/react-native-vision-camera #971: After install and yarn android Task :react-native-vision-camera:compileDebugKotlin FAILED

Heanphearak, 2022-04-07:
### Question

why error Task :react-native-vision-camera:compileDebugKotlin FAILED after follow document ?

### What I tried

after install react-native-vision-camera 
if configure follow document and [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting) but it still show me error like below : 
Task :react-native-vision-camera:compileDebugKotlin FAILED
w: Runtime JAR files in the classpath should have the same version. These files were found in the classpath:
    /Users/lucky/.gradle/caches/transforms-2/files-2.1/47cdc488023529fa897a4864ec26534b/jetified-kotlin-stdlib-jdk8-1.5.30.jar (version 1.5)
    /Users/lucky/.gradle/caches/transforms-2/files-2.1/94d1e5115193ad656111b17eef751d88/jetified-kotlin-stdlib-jdk7-1.5.30.jar (version 1.5)
    /Users/lucky/.gradle/caches/transforms-2/files-2.1/dc829bc1112a8836e30a6bf9905f68d9/jetified-kotlin-stdlib-1.6.10.jar (version 1.6)
    /Users/lucky/.gradle/caches/transforms-2/files-2.1/1179025ac1eeb681ddd6aa9e907cc93c/jetified-kotlin-stdlib-common-1.6.10.jar (version 1.6)
w: Some runtime JAR files in the classpath have an incompatible version. Consider removing them from the classpath
w: Detected multiple Kotlin daemon sessions at build/kotlin/sessions

### VisionCamera Version

"react-native-vision-camera": "^2.13.1",

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-04-07:
Hi.

This is not an error, you shared warnings. Each line starts with `w: ...` 🤦‍♂️ 

Also, why are you using the question template? This clearly is a bug not a question!

closed 2022-04-07 by mrousavy

ArjjunM, 2022-04-21:
Hi, 
Is there any solution for above mentioned issue, got the same error in android while taking build via jenkins, vision-camera version used is 2.13.1 and react-native version is 0.66.0.

 Execution failed for task ':react-native-vision-camera:compileDebugKotlin'

bekatd, 2022-04-26:
@frankbolviken please see my issue. I fixed it by downgrading to version 2.13.0

dunggnguyen, 2022-05-30:
Same problem, anyone have solution without downgrade

Vladislava9009, 2022-06-22:
The same problem

mangBert, 2022-07-24:
Same issue here. Is there a proper guide to solve this issue? Thanks
Build failed 
RN 0.65
vision camera version 2.13.0

RN 0.64.1
vision camera version 2.13.0

w: Runtime JAR files in the classpath should have the same version. These files were found in the classpath:
    /Users/albert/.gradle/caches/transforms-3/a51aa6ffd9c20d09211f194077ad9a53/transformed/jetified-kotlin-stdlib-jdk8-1.5.30.jar (version 1.5)
    /Users/albert/.gradle/caches/transforms-3/49a97541f1ac709a9bab3108a56c74cb/transformed/jetified-kotlin-stdlib-jdk7-1.5.30.jar (version 1.5)
    /Users/albert/.gradle/caches/transforms-3/304a1e5f9822cd8f57dcd5f6e020b1ef/transformed/jetified-kotlin-stdlib-1.6.10.jar (version 1.6)
    /Users/albert/.gradle/caches/transforms-3/b41570b21ffb26630fc89b122d7b7cff/transformed/jetified-kotlin-stdlib-common-1.6.10.jar (version 1.6)

kofkgoing, 2022-11-03:
```
e: /Users/young-jun/Projects/React-Native/KickGoing/Src/kickgoing-app/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (87, 116): Unresolved reference: resolveView
e: /Users/young-jun/Projects/React-Native/KickGoing/Src/kickgoing-app/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/FrameProcessorRuntimeManager.kt: (65, 76): Unresolved reference: resolveView

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> Compilation error. See log for more details
```

kofkgoing, 2022-11-03:
When I tried to create APK
```
e: ..../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (87, 116): Unresolved reference: resolveView
```

inkCrazy, 2022-11-22:
Same problem
react-native 0.63.5
react-native-vision-camera 2.13.0 or 2.15.2

mingkaili, 2022-11-26:
> Same problem react-native 0.63.5 react-native-vision-camera 2.13.0 or 2.15.2

+1, is there a solution for this yet?