margelo/react-native-vision-camera #973: 🐛 Video pausing does not work correctly on iOS

DragonSpirit, 2022-04-07:
### What were you trying to do?

Just call pauseRecording and resumeRecording() after start of recording

### Reproduceable Code

```tsx
It could be reproduced on example app
```

### What happened instead?

The video gets stuck on the last frame between pause and resume, without shortening the gap between them. Audio does not work correctly either.
Android works like a charm.

### Device

iPhone 11, iOS 15.4.1

### VisionCamera Version

2.13.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

itershukov, 2022-04-07:
Have the same issue on iPhone 7

itershukov, 2022-05-05:
Is there any news?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy