margelo/react-native-vision-camera #975: "Camera" API documentation not found 🐛

jenskuhrjorgensen, 2022-04-08:
### What were you trying to do?

Hi

I was trying to open the "Camera" link in this page https://mrousavy.com/react-native-vision-camera/docs/api but this link cannot be found https://mrousavy.com/react-native-vision-camera/docs/classes/Camera

Best regards
Jens

### What happened instead?

Page not found

### Device

Not relevant

### VisionCamera Version

Not relevant

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-08:
Huh, this works for me?

vgorte, 2022-04-08:
Where did you get this link?
 `https://mrousavy.com/react-native-vision-camera/docs/classes/Camera`

Navigating from `/docs/api` to the camera docs results in a different link:
 `https://mrousavy.com/react-native-vision-camera/docs/api/classes/Camera`

jenskuhrjorgensen, 2022-04-08:
Hmm, good question. I think I might have used the search functionality. Can you reproduce similar "not found" issues if you search for "torch" and click on the first result? That takes me to thus non-existing URL: https://mrousavy.com/react-native-vision-camera/docs/api/interfaces/cameraprops.cameraprops-1/#torch

[image]

mrousavy, 2022-04-11:
hmm, maybe this will get fixed if I upgrade docusaurus?

losh11, 2022-05-30:
The entire search part of the api website is broken for me. Not able to search for anything without the not found error.

mrousavy, 2022-05-31:
@losh11 Yeah I don't know why. If you have any idea how to fix it, please let me know.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy