margelo/react-native-vision-camera #977: 🐛 Unable to build app with boost-for-react-native

sanketkheni01, 2022-04-08:
### What were you trying to do?

Just developing the app.

### What happened instead?

[image]
This package downloads react-native boost each time. it takes so much time on each building. and also fails to build sometimes.

### Relevant log output

```shell
BUILD FAILED in 3m 5s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081
Note: I:\Just Share\Mobile App\node_modules\react-native-gradle-plugin\src\main\java\com\facebook\react\codegen\generator\SchemaJsonParser.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: I:\Just Share\Mobile App\node_modules\@react-native-async-storage\async-storage\android\src\main\java\com\reactnativecommunity\asyncstorage\AsyncStorageModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: I:\Just Share\Mobile App\node_modules\@react-native-async-storage\async-storage\android\src\main\java\com\reactnativecommunity\asyncstorage\AsyncStoragePackage.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: I:\Just Share\Mobile App\node_modules\react-native-push-notification\android\src\main\java\com\dieam\reactnativepushnotification\modules\RNPushNotification.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: I:\Just Share\Mobile App\android\app\src\main\java\com\reactnativeboilerplate\MainApplication.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug'.
> Build command failed.
  Error while executing process C:\Users\sanke\AppData\Local\Android\Sdk\cmake\3.10.2.4988404\bin\ninja.exe with arguments {-C I:\Just Share\Mobile App\node_modules\react-native-vision-camera\android\.cxx\Debug\542b6v5j\armeabi-v7a VisionCamera}
  ninja: Entering directory `I:\Just Share\Mobile App\node_modules\react-native-vision-camera\android\.cxx\Debug\542b6v5j\armeabi-v7a'
  [1/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o
  [2/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JImageProxy.cpp.o
  [3/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o
  [4/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o
  [5/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o
  [6/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/CameraView.cpp.o
  [7/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  C:\Users\sanke\AppData\Local\Android\Sdk\ndk\21.4.7075529\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe --target=armv7-none-linux-androideabi21 --gcc-toolchain=C:/Users/sanke/AppData/Local/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/sanke/AppData/Local/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/windows-x86_64/sysroot  -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -I"I:/Just Share/Mobile App/node_modules/react-native/React" -I"I:/Just Share/Mobile App/node_modules/react-native/React/Base" -I"I:/Just Share/Mobile App/node_modules/react-native/ReactAndroid/src/main/jni" -I"I:/Just Share/Mobile App/node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni" -I"I:/Just Share/Mobile App/node_modules/react-native/ReactCommon" -I"I:/Just Share/Mobile App/node_modules/react-native/ReactCommon/callinvoker" -I"I:/Just Share/Mobile App/node_modules/react-native/ReactCommon/jsi" -I"I:/Just Share/Mobile App/node_modules/hermes-engine/android/include" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/headers/AnimatedSensor" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/headers/Tools" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/headers/SpecTools" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/headers/Registries" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/headers/LayoutAnimations" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/hidden_headers" -I../../../../src/main/cpp -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_HAVE_MEMRCHR=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DON_ANDROID -DONANDROID -DFOR_HERMES=true -O0 -fno-limit-debug-info  -fPIC   -std=gnu++14 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -MF CMakeFiles\VisionCamera.dir\src\main\cpp\VisionCamera.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -c ../../../../src/main/cpp/VisionCamera.cpp
  In file included from ../../../../src/main/cpp/VisionCamera.cpp:3:
  In file included from ../../../../src/main/cpp/FrameProcessorRuntimeManager.h:14:
  In file included from ../../../../src/main/cpp/reanimated-headers/AndroidScheduler.h:8:
  In file included from I:/Just Share/Mobile App/node_modules/react-native/ReactAndroid/src/main/jni\react/jni/CxxModuleWrapper.h:10:
  In file included from I:/Just Share/Mobile App/node_modules/react-native/ReactAndroid/src/main/jni\react/jni/CxxModuleWrapperBase.h:13:
  In file included from I:/Just Share/Mobile App/node_modules/react-native/ReactCommon\cxxreact/CxxModule.h:15:
  ../../../../build/third-party-ndk/folly\folly/dynamic.h:63:10: fatal error: 'boost/operators.hpp' file not found
  #include <boost/operators.hpp>
           ^~~~~~~~~~~~~~~~~~~~~
  1 error generated.
  [8/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o
  C:\Users\sanke\AppData\Local\Android\Sdk\ndk\21.4.7075529\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe --target=armv7-none-linux-androideabi21 --gcc-toolchain=C:/Users/sanke/AppData/Local/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/sanke/AppData/Local/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/windows-x86_64/sysroot  -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -I"I:/Just Share/Mobile App/node_modules/react-native/React" -I"I:/Just Share/Mobile App/node_modules/react-native/React/Base" -I"I:/Just Share/Mobile App/node_modules/react-native/ReactAndroid/src/main/jni" -I"I:/Just Share/Mobile App/node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni" -I"I:/Just Share/Mobile App/node_modules/react-native/ReactCommon" -I"I:/Just Share/Mobile App/node_modules/react-native/ReactCommon/callinvoker" -I"I:/Just Share/Mobile App/node_modules/react-native/ReactCommon/jsi" -I"I:/Just Share/Mobile App/node_modules/hermes-engine/android/include" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/headers/AnimatedSensor" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/headers/Tools" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/headers/SpecTools" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems" -I"I:/Just Share/Mobile App/node_modules/react-native-reanimated/Common/cpp/headers/Registries" -I"I:/Just Share/Mobile App/node_modules/react-nat [... cut ...]
Android 12

### VisionCamera Version

2.13.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-11:
Hi!

This is not a bug, I have to download boost from somewhere. This will hopefully get fixed soon in the future by react-native core once they start providing prefabs.

closed 2022-04-11 by mrousavy

Jamal-Nazzal, 2022-04-14:
I've been facing this issue for a while, had to remove the camera library until a fix is done somewhere. why does this happen anyway? is it a bug with the boost library?

mrousavy, 2022-04-15:
Which RN version are you even on?

Jamal-Nazzal, 2022-04-15:
it's 0.67.3 and could you review my answers here please? they describe the issue well enough [https://github.com/mrousavy/react-native-vision-camera/issues/908](url)
thanks

Baira, 2022-04-19:
This happens sometimes when boost have not been installed properly. Just do this : 
- Delete node_modules folder
- gradlew clean
- yarn
- yarn react-native run-android

Jamal-Nazzal, 2022-04-19:
Thanks Mohamad, yeah I know the drill but this time it's different, every time boost does not install properly and you'd have to do these steps and even then the first build fails and you have to delete the build folder and build again. it's very tedious and there's something wrong here not sure where. this issue here explains the situation very clearly. #908

Jamal-Nazzal, 2022-04-26:
This issue hasn't been resolved for Windows yet

sanketkheni01, 2022-04-26:
@Baira Do we need to do this each time we encounter this issue?

sanketkheni01, 2022-04-26:
> This happens sometimes when boost have not been installed properly. Just do this : 
> - Delete node_modules folder
> - gradlew clean
> - yarn
> - yarn react-native run-android

Do we need to do this each time we encounter this issue?

Baira, 2022-04-26:
> Do we need to do this each time we encounter this issue?

Unfortunately yes. But once you resolve it, you should be good. I never encountered it since I deleted node_modules folder and rebuilt project.

Baira, 2022-04-29:
I just cloned and old project to start a new one and got the same error when building the app. Did `yarn add react-native-vision-camera react-native-reanimated` to update them to the latest version and everything worked fine without deleting `node_modules` or `gradlew clean`. The project is using `react-native v0.67.4`.

Hope this helps someone.

formifan2002, 2022-05-02:
see solution in: https://github.com/mrousavy/react-native-vision-camera/issues/908