margelo/react-native-vision-camera #978: error in example

safeblood, 2022-04-09:
### What were you trying to do?

yarn bootstrap

### Reproduceable Code

```tsx
yarn bootstrap
```

### What happened instead?

$ cd ios && bundle install
'bundle' is not recognized as an internal or external command,
operable program or batch file.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

### Relevant log output

```shell
$ cd ios && bundle install
'bundle' is not recognized as an internal or external command,
operable program or batch file.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

### Device

HP computer

### VisionCamera Version

any

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-10:
Building iOS apps does not work on Windows.

closed 2022-04-10 by mrousavy

safeblood, 2022-04-10:
> Building iOS apps does not work on Windows.

that's make sence.if i only want to build android project what should i do

mrousavy, 2022-04-11:
```sh
yarn
cd example
yarn
yarn android
```