margelo/react-native-vision-camera #979: Javascript worklet error 🐛

yamila-fraiman, 2022-04-09:
### What were you trying to do?

I want to implement a QR reader. It was working fine but suddenly I'm getting an error

### Reproduceable Code

```tsx
/* global scanQRCodes */
import type {Frame} from 'react-native-vision-camera';

export const scanQRCodes = (frame: Frame): string[] => {
	'worklet';
	// @ts-ignore
	return __scanQRCodes(frame);
};

const frameProcessor = useFrameProcessor(frame => {
		'worklet';
		const qrcode = scanQRCodes(frame);
		runOnJS(onBarcodesDetected)(qrcode);
	}, []);

module.exports = {
	presets: ['module:metro-react-native-babel-preset'],
	plugins: [
		'react-native-paper/babel',
		[
			'react-native-reanimated/plugin',
			{
				globals: ['__scanQRCodes'],
			},
		],
	],
};
```

### What happened instead?

Frame Processor threw an error: Exception in HostFunction: Javascript worklet error [native code]

### Device

Android 12

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-10:
I can't help you when the issue template is not filled out correctly.

Collect adb log output and paste it under "Relevant Log Output"

closed 2022-04-10 by mrousavy

LmRArtist, 2022-09-15:
i hva this error in IOS

bglgwyng, 2022-09-18:
I found this error after upgrading RN to 0.70.1.