margelo/react-native-vision-camera #980: 🐛 I try to open camera on IOS but it show blank screen ?

Heanphearak, 2022-04-11:
### What were you trying to do?

I try follow this : https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting

After that I run it on android device it working. but for ios it show blank screen when i click to open camera. 

I don't know .

### Reproduceable Code

```tsx
I use :
 
"react-native": "0.64.2",
"react-native-vision-camera": "2.13.0",
"react-native-reanimated": "2.2.4",
"vision-camera-code-scanner": "^0.2.0"
```

### What happened instead?

this is my code : 

import React, {useCallback} from 'react';

import {StyleSheet, Text} from 'react-native';
import {Camera, useCameraDevices} from 'react-native-vision-camera';
import {useScanBarcodes, BarcodeFormat} from 'vision-camera-code-scanner';

export default function App() {
  const [hasPermission, setHasPermission] = React.useState(false);
  const devices = useCameraDevices();
  const device = devices.back;

  console.log('===>devices', devices, device);

  const [frameProcessor, barcodes] = useScanBarcodes([BarcodeFormat.QR_CODE], {
    checkInverted: true,
  });

  React.useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'authorized');
    })();
  }, []);

  const onError = useCallback(error => {
    console.error('===>errr', error);
  }, []);

  console.log('MyPermission====>', hasPermission);

  console.log('my qr==>', barcodes);

  return (
    device != null &&
    hasPermission && (
      <>
        <Camera
          onError={onError}
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={true}
          frameProcessor={frameProcessor}
          frameProcessorFps={5}
        />
        {barcodes.map((barcode, idx) => (
          <Text key={idx} style={styles.barcodeTextURL}>
            {barcode.displayValue}
[... 839 lines cut ...]
2022-04-11 14:15:21.199839+0700 MorakotHR[2390:771256] [native] VisionCamera.configureCaptureSession(): Configuring Session...
2022-04-11 14:15:21.199906+0700 MorakotHR[2390:771256] [native] VisionCamera.configureCaptureSession(): Initializing Camera with device com.apple.avfoundation.avcapturedevice.built-in_video:0...
2022-04-11 14:15:21.199989+0700 MorakotHR[2390:771256] [native] VisionCamera.configureCaptureSession(): Adding Video input...
2022-04-11 14:15:21.200691+0700 MorakotHR[2390:771156] FrameProcessorBindings: Setting new frame processor...
2022-04-11 14:15:21.200743+0700 MorakotHR[2390:771156] FrameProcessorBindings: Adapting Shareable value from function (conversion to worklet)...
2022-04-11 14:15:21.201185+0700 MorakotHR[2390:771156] FrameProcessorBindings: Successfully created worklet!
2022-04-11 14:15:21.201272+0700 MorakotHR[2390:771302] FrameProcessorBindings: Converting worklet to Objective-C callback...
Assertion failed: (object->ctx_ == ctx_ && "Don't try to clone an object backed by a different Runtime"), function cloneObject, file JSCRuntime.cpp, line 611.
(lldb)
```

### Device

iphone xs max version 14.8.1

### VisionCamera Version

"react-native-vision-camera": "^2.13.1",

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-11:
This issue has been reported a couple of times before

> Assertion failed: (object->ctx_ == ctx_ && "Don't try to clone an object backed by a different Runtime"), function cloneObject, file JSCRuntime.cpp, line 611.
(lldb)

Heanphearak, 2022-04-11:
> This issue has been reported a couple of times before
> 
> > Assertion failed: (object->ctx_ == ctx_ && "Don't try to clone an object backed by a different Runtime"), function cloneObject, file JSCRuntime.cpp, line 611.
> > (lldb)

closed 2022-04-11 by Heanphearak

mrousavy, 2022-04-11:
It's because of your Frame Processor, not sure why though.

Heanphearak, 2022-04-11:
so how can i fix it ? 
I just start coding so don't understand much more to solve the issue. 
could you give me step by step to solve this issue.