margelo/react-native-vision-camera #992: 🐛 App crash in captureOutput

tinaszheng, 2022-04-13:
### What were you trying to do?

App crashes when camera scanner is mounted on the screen. This happens maybe 1/10 times, so not on every render.

### What happened instead?

App freezes with this error in xcode:

`Swift/arm64-apple-ios.swiftinterface:30564: Fatal error: Double value cannot be converted to UInt64 because it is either infinite or NaN`

on line 219 of `CameraView+RecordVideo.swit` 
[image]

### Device

iPhone 12

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-14:
Hey, thanks for the bug report! That's quite interesting, I wonder how it got there. Maybe division by 0?

gary-cohen, 2022-08-18:
I'm seeing the same issue consistently when I `runOnJS(setCameraAvailable)(true)` in the worklet where `setCameraAvailable` is the setting in useState.

schwrzd, 2023-06-11:
Had the same issue, this is my workaround:

```
  let nanosecondsPerFrame = secondsPerFrame * 1_000_000_000.0
        // Check if nanosecondsPerFrame is a valid number
        if !nanosecondsPerFrame.isNaN && !nanosecondsPerFrame.isInfinite {
            let nanosecondsPerFrameUInt64 = UInt64(nanosecondsPerFrame)
            if lastFrameProcessorCallElapsedTime >= nanosecondsPerFrameUInt64 {
        if !isRunningFrameProcessor {
          // we're not in the middle of executing the Frame Processor, so prepare for next call.
          CameraQueues.frameProcessorQueue.async {
            self.isRunningFrameProcessor = true

            let perfSample = self.frameProcessorPerformanceDataCollector.beginPerformanceSampleCollection()
            let frame = Frame(buffer: sampleBuffer, orientation: self.bufferOrientation)
            frameProcessor(frame)
            perfSample.endPerformanceSampleCollection()

            self.isRunningFrameProcessor = false
          }
          lastFrameProcessorCall = frameTime
        } else {
          // we're still in the middle of executing a Frame Processor for a previous frame, so a frame was dropped.
          ReactLogger.log(level: .warning, message: "The Frame Processor took so long to execute that a frame was dropped.")
        }
      }
    }
```

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy